victronenergy/venus #462: DVCC: improve "Limit charge current"

vejpasop, 2019-05-21:
### Summary: 

At present we have the following issues:
1. Switching the user setting ``Limit charge current`` from enable  to disable  leaves  the maximum charge current at the last user selected ``Maximum charge current`` value until the Multi is reset.
2. The maximum charge current can be set higher than the maximum charge current as configured through VEConfigure

Disabling ``Limit charge current`` should restore the original maximum charge current as configured through VEConfigure

### Details:
SystemCalc writes the maximum charge current to  ``Dc/0/MaxChargeCurrent``  (mk2-dbus)
Currently this path clamps the value to the maximum allowed value of the setting as reported by the Multi

### Todo:
**Mk2-dbus:**  

- [ ] Clamp ``Dc/0/MaxChargeCurrent`` to the EEPROM setting in the Multi. 

**SystemCalc:** 

- [ ] Write a high maximum charge current to ``Dc/0/MaxChargeCurrent`` when disabling "Limit charge current".  See also https://github.com/victronenergy/dbus-systemcalc-py/blob/master/delegates/dvcc.py#L722
- [ ] Limit the ``Maximum Charge Current`` user setting to the maximum value reported by ``Dc/0/MaxChargeCurrent``

vejpasop, 2019-05-21:
> Clamp Dc/0/MaxChargeCurrent to the EEPROM setting in the Multi.

Fixed in mk2-dbus v2.87

closed 2019-05-23 by jhofstee

mdkeil, 2023-09-30:
Doesnt work, I have to set Dc/0/MaxChargeCurrent manuelly to 70Amps again, after disabling "Limit Charge Current" via DVCC. 

using 3.20~7 .. same with 3.10