victronenergy/venus #464: Hiawatha SSL support

ezraholm50, 2019-05-26:
Hello!

I need to be able to access the remote console on LAN via HTTPS for compatibility reasons.

Now on my CCGX in SSH (ROOT) editing /etc/hiawatha/hiawatha.conf to look like this:

```
# GENERAL SETTINGS
#
#ServerId = www-data
ConnectionsTotal = 150
ConnectionsPerIP = 10
SystemLogfile = /var/volatile/log/hiawatha/system.log
GarbageLogfile = /var/volatile/log/hiawatha/garbage.log
ExploitLogfile = /var/volatile/log/hiawatha/exploit.log

# BINDING SETTINGS
# A binding is where a client can connect to.
#
Binding {
    Port = 80
    maxRequestSize = 512
#    Interface = 127.0.0.1
#    MaxKeepAlive = 30
#    TimeForRequest = 3,20
}
#
Binding {
    Port = 443
#    Interface = ::1
#    MaxKeepAlive = 30
#    TimeForRequest = 3,20
    SSLcertFile = /etc/hiawatha/cert.pem
}

# BANNING SETTINGS
# Deny service to clients who misbehave.
#
#BanOnGarbage = 300
#BanOnMaxPerIP = 60
#BanOnMaxReqSize = 300
#KickOnBan = yes
#RebanDuringBan = yes

# COMMON GATEWAY INTERFACE (CGI) SETTINGS
# These settings can be used to run CGI applications. Use the 'php-fcgi'
# tool to start PHP as a FastCGI daemon.
#
#CGIhandler = /usr/bin/perl:pl
#CGIhandler = /usr/bin/php-cgi:php
#CGIhandler = /usr/bin/python:py
#CGIhandler = /usr/bin/ruby:rb
#CGIhandler = /usr/bin/ssi-cgi:shtml
#CGIextension = cgi
#
#FastCGIserver {
#    FastCGIid = PHP5
#    ConnectTo = 127.0.0.1:2005
#    Extension = php
#}
FastCGIserver {
        FastCGIid = PHP5
        ConnectTo = /var/run/php5-fpm.socket
        Extension = php
}
UseFastCGI = PHP5
ExecuteCGI = yes

# URL TOOLKIT
# This URL toolkit rule was made for the Banshee PHP framework, which
# can be downloaded from http://www.hiawatha-webserver.org/banshee
#
#UrlToolkit {
#    ToolkitID = banshee
#    RequestURI isfile Return
#    Match ^/(css|files|images|js|slimstat)($|/) Return
#    Match ^/(favicon.ico|robots.txt|sitemap.xml)$ Return
#    Match .*\?(.*) Rewrite /index.php?$1
#    Match .* Rewrite /index.php
#}

# DEFAULT WEBSITE
# It is wise to use your IP address as the hostname of the default website
# and give it a blank webpage. By doing so, automated webscanners won't find
# your possible vulnerable website.
#
Hostname = 127.0.0.1
WebsiteRoot = /var/www/hiawatha
StartFile = index.html
AccessLogfile = none
ErrorLogfile = /var/volatile/log/hiawatha/error.log
#ErrorHandler = 404:/error.cgi

# VIRTUAL HOSTS
include /etc/hiawatha/sites-enabled

# Use a VirtualHost section to declare the websites you want to host.
#
#VirtualHost {
#    Hostname = www.my-domain.com
#    WebsiteRoot = /var/www/my-domain/public
#    StartFile = index.php
#    AccessLogfile = /var/www/my-domain/log/access.log
#    ErrorLogfile = /var/www/my-domain/log/error.log
#    TimeForCGI = 5
#    UseFastCGI = PHP5
#    UseToolkit = banshee
#}

# DIRECTORY SETTINGS
# You can specify some settings per directory.
#
#Directory {
#    Path = /home/baduser
#    ExecuteCGI = no
#    UploadSpeed = 10,2
#}
```

I've created a self signed cert (testing only for now) mentioned in the SSL setup.

https://www.hiawatha-webserver.org/howto/bindings

The issue is that the newest version 10.9 uses TLSCertFile instead of SSLCertFile. My Hiawatha version is 10.3 and should use TLSCertFile https://gitlab.com/hsleisink/hiawatha/blob/master/ChangeLog#L166. The generic config before I edited it was SSLCertFile.

Both wont work and give the following error:

```
root@ccgx:/etc/hiawatha# /etc/init.d/hiawatha restart
Restarting Hiawatha Web Server: no /usr/sbin/hiawatha found; none killed
Syntax error in hiawatha.conf on line 30.
hiawatha.
Where line 30 is the SSL line.
```

Please advise.

edit: how would i upgrade to 10.9 using opkg, can't seem to figure this out.

ezraholm50, 2019-09-03:
Where would i monitor if this would be merged in to a future release? Thanks
(As a work around i'm using stunnel and haprox to achieve SSL, with some changes to the /app source files)

mpvader, 2019-09-04:
Hi @ezraholm50; you can monitor this issue. I'll rename it to both updating to a newer hiawatha & looking into enabling SSL.

There might be caveats in enabling SSL (such as higher cpu usage), or not: it needs looking into.

I can't give you any timeline; sorry.

ezraholm50, 2019-09-06:
Thats great, id be happy to help test a beta!

DumpySquare, 2022-06-30:
I'd like to chime in for support of this feature.   Mainly to support some sort of security.  TLS, even for internal traffic, is the internet standard.  So much that big name browsers are looking to drop support for connecting to "insecure" sites completely.  Some tools and site won't even allow you to configure/setup without TSL AND some sort of authentication.

Overall, I think the best solution would be to move all to connectivity TLS with a self signed cert and setup a login.  It looks like most of that can be done within the native stack.   I know cpu usage was already mentioned, but were only talking about a few connections and it's already happening for cloud connectivity like VRM.  So the impact should be minimal.  Besides, I think security in this situation is critical.  This software runs critical systems for many people.  Hackers have been using any IOT device, router, or even baby monitors to do horrible things.  Again, this is a critical system for most, so security should be a priority.  

At a minimum add a setting to allow TLS offload and setup access lists so other technology, like NGINX can front end and provide all the security. 

I am happy to help.

mpvader, 2022-07-01:
Noted @DumpySquare , and thanks for pinging this issue. I'll see what we can do to improve it.

closed 2024-03-07 by jhofstee

DumpySquare, 2024-03-07:
> closing this since hiawatha has been replaced by nginx and has https support with a self signed certificate.

This is amazing news.  so happy.  thank you

WaaromZoMoeilijk, 2024-03-16:
No way!!!! that finally means time for iframes :smile:

jhofstee, 2024-03-17:
@WaaromZoMoeilijk Thanks, https://github.com/victronenergy/javascript-vnc-client/commit/2c6d1f33534d5dfd39feb7432be0cdab2ffad9f2 should fix that.

jhofstee, 2024-03-17:
This password over https part is included in Venus version v3.30~19 (or newer)