victronenergy/venus #465: call rc.local and rcS.local with start / stop arguments

SturmUDrang, 2019-05-29:
Hi!

for safely stopping custom services it would be nice if it worked like other startup scripts.

REF:  #169

mpvader, 2019-05-29:
Hi, you wrote:

> for safely stopping custom services it would be nice if it worked like other
> startup scripts.
> REF: https://github.com/victronenergy/venus/issues/169

I'm sorry, but I don't know what you mean. As far as I can see; it all works fine. Otherwise, perhaps put in a version of the code that you like better; and explain why. So send in a patch.

An idea on how the existing functionality can be used:  make the early started one of the rc scripts on /data to install daemontools  services on boot. So that is just like gui, vrmlogger, vedriect-interface and the rest.

And then you can control them with svc -d, svc -u etcetera.

Best regards, Matthijs

SturmUDrang, 2019-05-29:
Hi!

I would just pass on the start or stop parameter to rc.local for convenience:
`
 #!/bin/sh
if test -x /data/rc.local; then
   /data/rc.local $1
fi
exit 0
`

closed 2020-01-30 by jhofstee