victronenergy/venus #467: WiFi Failed status

mpvader, 2019-06-03:
As seen on system `Test Venus Eco 13226` of IMV:

```
May 29 11:37:12 beaglebone daemon.notice wpa_supplicant[1514]: wlan0: Trying to associate with e0:46:9a:3d:92:3b (SSID='WirelessIMV' freq=2462 MHz)
May 29 11:37:12 beaglebone user.info kernel: [31241.482042] rt2800usb 2-1.1:1.0 wlan0: disabling HT/VHT due to WEP/TKIP use
May 29 11:37:12 beaglebone user.info kernel: [31241.490263] wlan0: associate with e0:46:9a:3d:92:3b (try 1/3)
May 29 11:37:12 beaglebone daemon.warn connmand[1454]: Probably roaming right now! Staying connected...
May 29 11:37:12 beaglebone user.info kernel: [31241.503156] wlan0: RX AssocResp from e0:46:9a:3d:92:3b (capab=0x531 status=0 aid=1)
May 29 11:37:12 beaglebone daemon.notice wpa_supplicant[1514]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
May 29 11:37:12 beaglebone user.info kernel: [31241.536482] wlan0: associated
May 29 11:37:12 beaglebone daemon.info connmand[1454]: wlan0 {RX} 108938 packets 14784640 bytes
May 29 11:37:12 beaglebone daemon.info connmand[1454]: wlan0 {TX} 19372 packets 3260637 bytes
May 29 11:37:12 beaglebone daemon.info connmand[1454]: wlan0 {update} flags 102403 <UP,LOWER_UP>
May 29 11:37:12 beaglebone daemon.info connmand[1454]: wlan0 {newlink} index 7 address 00:0F:00:35:BC:1E mtu 1450
May 29 11:37:12 beaglebone daemon.info connmand[1454]: wlan0 {newlink} index 7 operstate 5 <DORMANT>
May 29 11:37:12 beaglebone daemon.notice wpa_supplicant[1514]: wlan0: Associated with e0:46:9a:3d:92:3b
May 29 11:37:12 beaglebone daemon.notice wpa_supplicant[1514]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
May 29 11:37:12 beaglebone daemon.notice wpa_supplicant[1514]: wlan0: WPA: Key negotiation completed with e0:46:9a:3d:92:3b [PTK=TKIP GTK=TKIP]
May 29 11:37:12 beaglebone daemon.notice wpa_supplicant[1514]: wlan0: CTRL-EVENT-CONNECTED - Connection to e0:46:9a:3d:92:3b completed [id=0 id_str=]
May 29 11:37:12 beaglebone daemon.info avahi-daemon[1435]: Joining mDNS multicast group on interface wlan0.IPv6 with address fe80::20f:ff:fe35:bc1e.
May 29 11:37:12 beaglebone daemon.info avahi-daemon[1435]: New relevant interface wlan0.IPv6 for mDNS.
May 29 11:37:12 beaglebone daemon.info avahi-daemon[1435]: Joining mDNS multicast group on interface wlan0.IPv4 with address 172.19.5.36.
May 29 11:37:12 beaglebone daemon.info avahi-daemon[1435]: New relevant interface wlan0.IPv4 for mDNS.
May 29 11:37:12 beaglebone daemon.info avahi-daemon[1435]: Registering new address record for fe80::20f:ff:fe35:bc1e on wlan0.*.
May 29 11:37:12 beaglebone daemon.info avahi-daemon[1435]: Registering new address record for 172.19.5.36 on wlan0.IPv4.
May 29 11:37:12 beaglebone daemon.info connmand[1454]: wlan0 {RX} 108940 packets 14784922 bytes
May 29 11:37:12 beaglebone daemon.info connmand[1454]: wlan0 {TX} 19375 packets 3261060 bytes
May 29 11:37:12 beaglebone daemon.info connmand[1454]: wlan0 {update} flags 102467 <UP,RUNNING,LOWER_UP>
May 29 11:37:12 beaglebone daemon.info connmand[1454]: wlan0 {newlink} index 7 address 00:0F:00:35:BC:1E mtu 1450
May 29 11:37:12 beaglebone daemon.info connmand[1454]: wlan0 {newlink} index 7 operstate 6 <UP>
May 29 11:37:12 beaglebone daemon.warn connmand[1454]: Probably roaming right now! Staying connected...
May 29 11:37:12 beaglebone daemon.notice wpa_supplicant[1514]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=GB
May 29 11:37:13 beaglebone daemon.notice wpa_supplicant[1514]: wlan0: CTRL-EVENT-SIGNAL-CHANGE above=0 signal=-65 noise=9999 txrate=18000
May 29 11:38:56 beaglebone daemon.notice wpa_supplicant[1514]: wlan0: SME: Trying to authenticate with 00:1d:aa:12:f5:80 (SSID='WirelessIMV' freq=2437 MHz)
May 29 11:38:56 beaglebone user.info kernel: [31345.836008] wlan0: disconnect from AP e0:46:9a:3d:92:3b for new auth to 00:1d:aa:12:f5:80
May 29 11:38:57 beaglebone user.info kernel: [31345.892518] wlan0: authenticate with 00:1d:aa:12:f5:80
May 29 11:38:57 beaglebone daemon.info connmand[1454]: wlan0 {RX} 109495 packets 14856628 bytes
May 29 11:38:57 beaglebone daemon.info connmand[1454]: wlan0 {TX} 19441 packets 3278098 bytes
May 29 11:38:57 beaglebone daemon.info connmand[1454]: wlan0 {update} flags 36867 <UP>
May 29 11:38:57 beaglebone daemon.info avahi-daemon[1435]: Interface wlan0.IPv6 no longer relevant for mDNS.
May 29 11:38:57 beaglebone daemon.info avahi-daemon[1435]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::20f:ff:fe35:bc1e.
May 29 11:38:57 beaglebone daemon.info avahi-daemon[1435]: Interface wlan0.IPv4 no longer relevant for mDNS.
May 29 11:38:57 beaglebone daemon.info avahi-daemon[1435]: Leaving mDNS multicast group on interface wlan0.IPv4 with address 172.19.5.36.
May 29 11:38:57 beaglebone daemon.info avahi-daemon[1435]: Withdrawing address record for fe80::20f:ff:fe35:bc1e on wlan0.
May 29 11:38:57 beaglebone daemon.info avahi-daemon[1435]: Withdrawing address record for 172.19.5.36 on wlan0.
May 29 11:38:57 beaglebone user.info kernel: [31345.932233] wlan0: send auth to 00:1d:aa:12:f5:80 (try 1/3)
May 29 11:38:57 beaglebone user.info kernel: [31345.947143] wlan0: authenticated
May 29 11:38:57 beaglebone daemon.info connmand[1454]: wlan0 {newlink} index 7 address 00:0F:00:35:BC:1E mtu 1450
May 29 11:38:57 beaglebone daemon.info connmand[1454]: wlan0 {newlink} index 7 operstate 2 <DOWN>
May 29 11:38:57 beaglebone daemon.warn connmand[1454]: Probably roaming right now! Staying connected...
May 29 11:38:57 beaglebone daemon.notice wpa_supplicant[1514]: wlan0: Trying to associate with 00:1d:aa:12:f5:80 (SSID='WirelessIMV' freq=2437 MHz)
May 29 11:38:57 beaglebone user.info kernel: [31345.992405] wlan0: associating with AP with corrupt probe response
May 29 11:38:57 beaglebone user.info kernel: [31346.000883] wlan0: associate with 00:1d:aa:12:f5:80 (try 1/3)
May 29 11:38:57 beaglebone daemon.warn connmand[1454]: Probably roaming right now! Staying connected...
May 29 11:38:57 beaglebone daemon.notice wpa_supplicant[1514]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
May 29 11:38:57 beaglebone user.info kernel: [31346.129865] wlan0: associate with 00:1d:aa:12:f5:80 (try 2/3)
May 29 11:38:57 beaglebone user.info kernel: [31346.242287] wlan0: associate with 00:1d:aa:12:f5:80 (try 3/3)
May 29 11:38:57 beaglebone user.info kernel: [31346.379790] wlan0: association with 00:1d:aa:12:f5:80 timed out
May 29 11:38:57 beaglebone daemon.err connmand[1454]: Time request for server 172.19.4.1 failed (101/Network is unreachable)
May 29 11:38:57 beaglebone daemon.err connmand[1454]: Time request for server 172.19.4.1 failed (101/Network is unreachable)
May 29 11:38:57 beaglebone user.info kernel: [31346.468620] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
May 29 11:38:57 beaglebone daemon.info connmand[1454]: wlan0 {del} address 172.19.5.36/22 label wlan0
May 29 11:38:57 beaglebone daemon.info connmand[1454]: wlan0 {del} route 172.19.4.0 gw 0.0.0.0 scope 253 <LINK>
May 29 11:38:57 beaglebone daemon.info connmand[1454]: wlan0 {del} route fe80:: gw :: scope 0 <UNIVERSE>
May 29 11:38:57 beaglebone daemon.info connmand[1454]: wlan0 {del} route ff00:: gw :: scope 0 <UNIVERSE>
May 29 11:38:57 beaglebone daemon.info dnsmasq[1413]: stopped using nameserver 172.19.4.20#53
May 29 11:38:57 beaglebone daemon.notice wpa_supplicant[1514]: wlan0: SME: Trying to authenticate with e0:46:9a:3d:92:3b (SSID='WirelessIMV' freq=2462 MHz)
May 29 11:38:57 beaglebone user.info kernel: [31346.604270] wlan0: authenticate with e0:46:9a:3d:92:3b
May 29 11:38:57 beaglebone user.info kernel: [31346.672415] wlan0: send auth to e0:46:9a:3d:92:3b (try 1/3)
May 29 11:38:57 beaglebone user.info kernel: [31346.682111] wlan0: authenticated
[... 64 lines cut ...]
Jun  3 12:11:49 beaglebone daemon.info connmand[1454]: wlan0 {add} route ff00:: gw :: scope 0 <UNIVERSE>
Jun  3 12:11:49 beaglebone daemon.info connmand[1454]: wlan0 {add} route fe80:: gw :: scope 0 <UNIVERSE>
Jun  3 12:11:49 beaglebone daemon.info connmand[1454]: wlan0 {RX} 109497 packets 14856934 bytes
Jun  3 12:11:49 beaglebone daemon.info connmand[1454]: wlan0 {TX} 19443 packets 3278386 bytes
Jun  3 12:11:49 beaglebone daemon.info connmand[1454]: wlan0 {update} flags 102467 <UP,RUNNING,LOWER_UP>
Jun  3 12:11:49 beaglebone daemon.info connmand[1454]: wlan0 {newlink} index 7 address 00:0F:00:35:BC:1E mtu 1450
Jun  3 12:11:49 beaglebone daemon.info connmand[1454]: wlan0 {newlink} index 7 operstate 6 <UP>
[... cut ...]
Jun  3 12:11:50 beaglebone daemon.info avahi-daemon[1435]: Registering new address record for fe80::20f:ff:fe35:bc1e on wlan0.*.
Jun  3 12:11:55 beaglebone daemon.notice wpa_supplicant[1514]: wlan0: CTRL-EVENT-SIGNAL-CHANGE above=0 signal=-59 noise=9999 txrate=1000
Jun  3 12:11:55 beaglebone daemon.info connmand[1454]: wlan0 {add} route 212.227.81.55 gw 172.19.4.1 scope 0 <UNIVERSE>
Jun  3 12:11:57 beaglebone daemon.info connmand[1454]: wlan0 {del} route 212.227.81.55 gw 172.19.4.1 scope 0 <UNIVERSE>
Jun  3 12:12:27 beaglebone daemon.info connmand[1454]: ntp: time slew -20.937457 s
```

mpvader, 2019-06-03:
drmler site shows the same or similar in the logs:

Failure:
```
May 30 10:07:26 ccgx daemon.info connmand[808]: ntp: time slew +0.060719 s
May 30 10:11:52 ccgx daemon.notice wpa_supplicant[856]: wlan0: Reject scan trigger since one is already pending
May 30 10:16:52 ccgx daemon.notice wpa_supplicant[856]: wlan0: Reject scan trigger since one is already pending
May 30 10:18:26 ccgx user.info kernel: [113730.822408] wlan0: AP b4:86:55:87:7e:a0 changed bandwidth, new config is 2442 MHz, width 2 (2452/0 MHz)
May 30 10:21:52 ccgx daemon.notice wpa_supplicant[856]: wlan0: Reject scan trigger since one is already pending
May 30 10:21:55 ccgx user.info kernel: [113940.093050] wlan0: AP b4:86:55:87:7e:a0 changed bandwidth, new config is 2442 MHz, width 1 (2442/0 MHz)
May 30 10:26:52 ccgx daemon.notice wpa_supplicant[856]: wlan0: Reject scan trigger since one is already pending
May 30 10:31:52 ccgx daemon.notice wpa_supplicant[856]: wlan0: Reject scan trigger since one is already pending
May 30 10:36:52 ccgx daemon.notice wpa_supplicant[856]: wlan0: Reject scan trigger since one is already pending
May 30 10:41:52 ccgx daemon.notice wpa_supplicant[856]: wlan0: Reject scan trigger since one is already pending
May 30 10:46:52 ccgx daemon.notice wpa_supplicant[856]: wlan0: Reject scan trigger since one is already pending
May 30 10:51:52 ccgx daemon.notice wpa_supplicant[856]: wlan0: Reject scan trigger since one is already pending
May 30 10:56:52 ccgx daemon.notice wpa_supplicant[856]: wlan0: Reject scan trigger since one is already pending
May 30 11:01:52 ccgx daemon.notice wpa_supplicant[856]: wlan0: Reject scan trigger since one is already pending
May 30 11:06:52 ccgx daemon.notice wpa_supplicant[856]: wlan0: Reject scan trigger since one is already pending
May 30 11:11:52 ccgx daemon.notice wpa_supplicant[856]: wlan0: Reject scan trigger since one is already pending
May 30 11:16:52 ccgx daemon.notice wpa_supplicant[856]: wlan0: Reject scan trigger since one is already pending
May 30 11:18:03 ccgx daemon.info connmand[808]: wlan0 {RX} 116537 packets 10465934 bytes
May 30 11:18:03 ccgx daemon.info avahi-daemon[788]: Interface wlan0.IPv6 no longer relevant for mDNS.
May 30 11:18:03 ccgx daemon.info avahi-daemon[788]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::823f:5dff:fe22:678.
May 30 11:18:03 ccgx daemon.info avahi-daemon[788]: Interface wlan0.IPv4 no longer relevant for mDNS.
May 30 11:18:03 ccgx daemon.info avahi-daemon[788]: Leaving mDNS multicast group on interface wlan0.IPv4 with address 192.168.8.103.
May 30 11:18:03 ccgx daemon.info avahi-daemon[788]: Withdrawing address record for fe80::823f:5dff:fe22:678 on wlan0.
May 30 11:18:03 ccgx daemon.info avahi-daemon[788]: Withdrawing address record for 192.168.8.103 on wlan0.
May 30 11:18:03 ccgx user.info kernel: [117308.167120] wlan0: failed to use reserved channel context, disconnecting (err=-95)
May 30 11:18:03 ccgx user.info kernel: [117308.167472] wlan0: failed to finalize channel switch, disconnecting
May 30 11:18:03 ccgx daemon.info connmand[808]: wlan0 {TX} 52861 packets 12629637 bytes
May 30 11:18:03 ccgx daemon.info connmand[808]: wlan0 {update} flags 36867 <UP>
May 30 11:18:03 ccgx daemon.info connmand[808]: wlan0 {newlink} index 4 address 80:3F:5D:22:06:78 mtu 1450
May 30 11:18:03 ccgx daemon.info connmand[808]: wlan0 {newlink} index 4 operstate 2 <DOWN>
May 30 11:18:03 ccgx daemon.notice wpa_supplicant[856]: wlan0: CTRL-EVENT-DISCONNECTED bssid=b4:86:55:87:7e:a0 reason=4 locally_generated=1
May 30 11:18:03 ccgx daemon.err connmand[808]: Time request for server 192.168.8.1 failed (101/Network is unreachable)
May 30 11:18:03 ccgx daemon.err connmand[808]: Time request for server 192.168.8.1 failed (101/Network is unreachable)
May 30 11:18:03 ccgx user.info kernel: [117308.501319] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
May 30 11:18:03 ccgx daemon.info connmand[808]: wlan0 {del} address 192.168.8.103/24 label wlan0
May 30 11:18:03 ccgx daemon.info connmand[808]: wlan0 {del} route 192.168.8.0 gw 0.0.0.0 scope 253 <LINK>
May 30 11:18:03 ccgx daemon.info connmand[808]: wlan0 {del} route fe80:: gw :: scope 0 <UNIVERSE>
May 30 11:18:03 ccgx daemon.info connmand[808]: wlan0 {del} route ff00:: gw :: scope 0 <UNIVERSE>
May 30 11:18:03 ccgx daemon.notice wpa_supplicant[856]: wlan0: Reject scan trigger since one is already pending
May 30 11:18:04 ccgx daemon.notice wpa_supplicant[856]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
May 30 11:18:04 ccgx daemon.warn connmand[808]: Skipping disconnect of 57694669_managed_psk, network is connecting.
May 30 11:18:04 ccgx daemon.info dnsmasq[770]: stopped using nameserver 192.168.8.1#53
May 30 11:18:05 ccgx daemon.notice wpa_supplicant[856]: wlan0: SME: Trying to authenticate with b4:86:55:87:7e:a0 (SSID='WiFi' freq=2442 MHz)
May 30 11:18:05 ccgx user.info kernel: [117310.481483] wlan0: authenticate with b4:86:55:87:7e:a0
May 30 11:18:05 ccgx user.info kernel: [117310.591630] wlan0: send auth to b4:86:55:87:7e:a0 (try 1/3)
May 30 11:18:05 ccgx user.info kernel: [117310.632505] wlan0: send auth to b4:86:55:87:7e:a0 (try 2/3)
May 30 11:18:05 ccgx user.info kernel: [117310.662624] wlan0: send auth to b4:86:55:87:7e:a0 (try 3/3)
May 30 11:18:06 ccgx user.info kernel: [117310.692782] wlan0: authentication with b4:86:55:87:7e:a0 timed out
May 30 11:18:06 ccgx daemon.warn connmand[808]: Skipping disconnect of 57694669_managed_psk, network is connecting.
[... cut ...]
Jun  3 18:54:57 ccgx user.info kernel: [490322.508707] wlan0: send auth to b4:86:55:87:7e:a0 (try 1/3)
Jun  3 18:54:57 ccgx user.info kernel: [490322.511013] wlan0: authenticated
Jun  3 18:54:57 ccgx daemon.notice wpa_supplicant[856]: wlan0: Trying to associate with b4:86:55:87:7e:a0 (SSID='WiFi' freq=2452 MHz)
Jun  3 18:54:57 ccgx user.info kernel: [490322.522987] wlan0: associate with b4:86:55:87:7e:a0 (try 1/3)
Jun  3 18:54:57 ccgx user.info kernel: [490322.527808] wlan0: RX AssocResp from b4:86:55:87:7e:a0 (capab=0x1411 status=0 aid=4)
```

ThiemoVanEngelen, 2019-06-04:
The default behavior for a wifi service in failure is that connman will not try to reconnect. A service can leave the failure state due to 2 things: 
  1. There is a connect request via the dbus. 
  2. The service is removed because wpa_supplicant no longer offers the service.

Nr 1 is what happens when the user presses connect in the Venus gui.
Nr 2 happens when wpa_supplicant did not see the service for more than 3 minutes. This only happens when connman did not start a scan for 3 minutes. The default behavior of connman (while not connected) is to start a scan, then after 30 seconds a new scan and then connman continues to double the interval upto 5 minutes (again while it is not connected). So at one point, the interval between 2 scans initiated by connman is more than 3 minutes, causing wpa_supplicant to not see the service in a scan for more than 3 minutes at which point it is removed by wpa_supplicant. This is seen by connman, which also removes its the service, removing the failure state. At the next scan, wpa_supplicant will see the service again and publish it on the dbus, connman sees this and recreates the service and tries to reconnect (because it is no longer in the failure state).

So the whole story above with wpa_supplicant removing the service will not happen when the gui is in the wifi menu, as the wifi menu will start a scan every 10 seconds and wpa_supplicant will thus see the service always within 3 minutes.

ThiemoVanEngelen, 2019-06-04:
[Current venus connman patches](https://github.com/victronenergy/meta-victronenergy/tree/231be0423740fd5a0282f3b8a6ae0d907a3d6ed2/meta-venus/recipes-connectivity/connman/connman) change the wifi behavior in the following way:
When it tries to connect to a wifi service and gets an error that looks like the wifi password is wrong, it continuously tries to reconnect, due to the `FavoriteMaxRetries=-1` in the main.conf. This was done because also a marginal wifi reception might lead to connman thinking that the wifi password was wrong.

closed 2019-06-05 by mpvader