victronenergy/venus #469: BYD CANbus BMS timeouts?

mpvader, 2019-06-06:
Seems to cause no functional problems. And might be exemplary (is that English?) to that one system. First thing to do is check on other BYD systems.

https://community.victronenergy.com/questions/19389/venus-mqtt-bms-json-data-errors.html?childToView=20502#comment-20502

Hobohome, 2019-08-27:
I have a system setup that logs the occurrence of these disconnects. Seems to be fairly random and happening about 5 - 8 times each day. While there are no functional problems, it does cause a lot of issue with my monitoring system each time the BYD system disconnects from the venus.   
I am happy to test any proposed fixes on my system.

jhofstee, 2019-10-29:
BYD is horrible slow, normal update intervals are 1sec, 10 seconds for slow things.

mpvader, 2019-11-04:
Izak will look into this. Pls also report here what frames are sent at what interval.

It should be such that it can survive one missed packet. (even though canbus normally shouldn't ever miss a packet).

izak, 2019-11-11:
* Sent every 2 seconds: `0x351` - Charge voltage, Charge current, Discharge voltage, Discharge current
* Sent every 60 seconds: `0x35E` - Manufacturer name
* Everything else is sent every 10 seconds

`bmsPresent` is called when 0x351 is observed (and in a few other places), for our purposes here we can conclude that is should be called every 2 seconds.

`bmsPresent` must be called every 3 seconds otherwise the battery service disappears. So one "lost" frame is enough to cause the service to disappear for a second.

If we increase the timeout to 5 seconds, it should be enough to allow for one lost packet.

jhofstee, 2019-12-18:
I had a look at this:
- The timeout was anywhere between 2 to 3 seconds. If the timeout is reset just before the counter decrement it will be 2 seconds.
- A second lasts 1050 ms in v2.33, 1000ms in v2.40. So there is by accident a tiny margin in v2.33.
- The increment to 5 is actually anything between 4 and 5 seconds. It suffers the same issue if there were any lost messages (which I haven't seen).

So the change will likely fix the disconnect, but for other reasons then mentioned. I will make a proper version.

jhofstee, 2019-12-18:
For completeness, the root cause for this is that we expect batteries to update currents at least once a second or more often. As already mentioned BYD is annoyingly slow. That should be fixed as well..

jhofstee, 2019-12-18:
Even more oddly, for BYD we seem to force to be on, DVCC ain't going to work in systems with fluctuating loads, since there is a 10 sec delay, if anything we should turn it off I guess.

jhofstee, 2020-12-03:
This included quite some time ago.

closed 2020-12-03 by jhofstee