victronenergy/venus #472: mqtt-rpc: VE.Bus firmware versions to device-list

thlassche, 2019-06-18:
- [ ] Add VE.Bus devices to the device-list call with their firmware version and with a flag indicating that they can't be updated.
- [ ] Check that VRM shows them as not updatable with their firmware version in the device list

The reason to do this, is that people don't understand that not every class of devices is updatable through mqtt-rpc (https://github.com/victronenergy/vrm-portal/issues/119)

*Note:* It was also discussed that it is an option to show the Venus device itself in this list. It was decided that that was not necessary for now, but might become relevant again in the future (e.g. for updating the Venus device itself through mqtt-rpc)

thlassche, 2019-06-18:
/cc @mpvader. See also https://victrondevelopment.slack.com/archives/C8LV2EU94/p1558533146036400

mpvader, 2019-11-04:
@thlassche is this one we can finish up and close next wednesday and thursday? would be nice.

thlassche, 2019-11-04:
@mpvader Sure! I'll dive into it before wednesday so I'm prepared

mpvader, 2019-11-28:
Whats this waiting for, @thlassche ?

thlassche, 2019-11-28:
@mpvader Testing (which I just did on Hub-1) and a capability check, , already going quite well:

[image]

Let's quickly go through the remaining steps together tomorrow than I can wrap it up

mpvader, 2019-11-29:
As discussed just now: wrt planning this is not for v2.40; but hopefully we can include it in v2.50. Few small more changes are necessary; see above updated todo.

jhofstee, 2021-06-21:
Closing this, since it included quite some time ago.

closed 2021-06-21 by jhofstee