victronenergy/venus #474: Improve kWh handling around the BMV and has dc system

mpvader, 2019-06-21:
- [ ] (Thiemo) make vedirect_interface hide the `/History/DischargedEnergy` & `/History/ChargedEnergy` paths when productid in (bmv productids) and firmware <= 3.07
- [ ] (Matthijs) improve [the flawed workaround](https://git.victronenergy.com/ccgx/dbus_vrm/blob/master/kwhdeltas.py#L384) in vrmlogger

mpvader, 2019-06-21:
note to self: there already is a filter in kwhdeltas it ignore battery services that have known issues in their kWh calculation:

commit [kwhdeltas: ignore battery monitors with no or faulty kwhcounters](https://git.victronenergy.com/ccgx/dbus_vrm/commit/eda2a5091e6cbd917364b415e8edf7a8e9e73495)

ThiemoVanEngelen, 2024-02-26:
@mpvader Is this still necessary / desired in vedirect-interface as there is already a workaround in vrmlogger?