victronenergy/venus #475: mqtt-rpc: add more data to VregLink get-device-list

laknorbert, 2019-06-21:
For context, see get-device-list response for Groningen Hub-1 below.

VictronConnect needs to show an icon representative for the device, a friendly name and the model name.

Now, VictronConnect needs to go and query that for each device that it wants to show, the current implementation does so by calling GetVreg a few times.

It would be much easier if the productid, friendly name and model name are all in the get-device-list response.

get-device-list spec is [here](https://github.com/victronenergy/mqtt-rpc/blob/master/README.md#vreg-device-list).

Looks like there are two ways to do this:

1. Have all the drivers add more information to the /Devices/ subtree on D-Bus, and then current implementation of mqtt-rpc will already add it; as can be seen in below json.

2. Make mqtt-rpc a little bit smarter; and add productid, friendly name, model-name and possibly some more default information such as firmware version into the right (!) item in the device list.

```
{
   "devices":[
      {
         "vregLink":"no",
         "deviceInstance":257,
         "data":{

         },
         "id":"com.victronenergy.vebus.ttyO1/Devices/Dmc"
      },
      {
         "vregLink":"no",
         "deviceInstance":257,
         "data":{
            "ExtendStatus/GridRelayReport/Reset":0,
            "Version":2612459,
            "InterfaceProtectionLog/0/Time":1560823980,
            "InterfaceProtectionLog/0/ErrorFlags":8
         },
         "id":"com.victronenergy.vebus.ttyO1/Devices/0"
      },
      {
         "vregLink":"no",
         "deviceInstance":257,
         "data":{

         },
         "id":"com.victronenergy.vebus.ttyO1/Devices/Bms"
      },
      {
         "vregLink":"yes",
         "deviceInstance":258,
         "data":{

         },
         "id":"com.victronenergy.solarcharger.ttyO2/Devices/0"
      },
      {
         "vregLink":"yes",
         "deviceInstance":256,
         "data":{

         },
         "id":"com.victronenergy.battery.ttyO0/Devices/0"
      }
   ]
}
```

mpvader, 2019-06-21:
Wrt whats available on the canbus: see below readout (also from Groningen Hub1)

Productid is missing, and friendly name aka custom name.

```
# dbus -y com.victronenergy.vecan.can0 / GetValue
value = {
 'Devices/2CC00064/FirmwareVersion': '1.0',
 'Devices/2CC00064/Instance': 0,
 'Devices/2CC00064/Manufacturer': 358,
 'Devices/2CC00064/ModelName': 'VE.Can interface',
 'Devices/2CC00064/Nad': 1,
 'Devices/2CC00064/Serial': 'serial#',
 'Devices/2CC034F4/FirmwareVersion': '2.01',
 'Devices/2CC034F4/Instance': 6,
 'Devices/2CC034F4/Manufacturer': 358,
 'Devices/2CC034F4/ModelName': 'Skylla-i Control Panel',
 'Devices/2CC034F4/Nad': 183,
 'Devices/2CC034F4/Serial': '0013556 HQ1229VKPF9',
 'Devices/2CC0353F/FirmwareVersion': '2.01',
 'Devices/2CC0353F/Instance': 2,
 'Devices/2CC0353F/Manufacturer': 358,
 'Devices/2CC0353F/ModelName': 'Skylla-i Control Panel',
 'Devices/2CC0353F/Nad': 180,
 'Devices/2CC0353F/Serial': '0013631 HQ1229ZVH4Y',
 'Devices/2CC03D17/FirmwareVersion': '2.06',
 'Devices/2CC03D17/Instance': 2,
 'Devices/2CC03D17/Manufacturer': 358,
 'Devices/2CC03D17/ModelName': 'Skylla-i 24V/100A/3 outputs',
 'Devices/2CC03D17/Nad': 166,
 'Devices/2CC03D17/Serial': '0015639 HQ12424SLBZ',
 'Devices/2CC03D1C/FirmwareVersion': '2.06',
 'Devices/2CC03D1C/Instance': 2,
 'Devices/2CC03D1C/Manufacturer': 358,
 'Devices/2CC03D1C/ModelName': 'Skylla-i 24V/100A/3 outputs',
 'Devices/2CC03D1C/Nad': 59,
 'Devices/2CC03D1C/Serial': '0015644 HQ1242WUBN3',
 'Devices/2CC03E5D/FirmwareVersion': '2.05',
 'Devices/2CC03E5D/Instance': 2,
 'Devices/2CC03E5D/Manufacturer': 358,
 'Devices/2CC03E5D/ModelName': 'BlueSolar Charger MPPT 150/70',
 'Devices/2CC03E5D/Nad': 161,
 'Devices/2CC03E5D/Serial': '0015965 HQ1250944QQ',
 'Devices/2CC03F7C/FirmwareVersion': '2.05',
 'Devices/2CC03F7C/Instance': 1,
 'Devices/2CC03F7C/Manufacturer': 358,
 'Devices/2CC03F7C/ModelName': 'BlueSolar Charger MPPT 150/70',
 'Devices/2CC03F7C/Nad': 60,
 'Devices/2CC03F7C/Serial': '0016252 HQ1250JFX2W',
 'Link/ChargeVoltage': []
}
```

mpvader, 2019-06-21:
A summary of what was discussed in Slack:

1. The paths ProductId, ProductName, CustomName, FirmwareVersion will be added to the entries in the `/Devices` sub-tree on D-Bus. The reasons to do so is that for the end-user on VictronConnect, this will make for a more smooth experience. And also for devices that don't support VREGs (like VE.Bus) VictronConnect can still show them; with a proper icon; even though its main page and/or its settings can't be accessed.

2. Also, the DeviceInstance and ServiceName will be added to the entries in the `/Devices` sub-tree. Both will be used by VictronConnect to map from this list to the mqtt stream coming from dbus-mqtt; which is used to receive the asynchronous data. dbus-mqtt publishes data as `N/<portal ID>/<service_type>/<device instance>/<D-Bus path>`.

3. [The mqtt-rpc commit that does something with /DeviceInstance](https://github.com/victronenergy/mqtt-rpc/commit/7c5fc16356b818ab1731e02faa19962a7bd5b093) can be reverted; since the DeviceInstance will automatically be added to the `Data` part in the dict. And also using /DeviceInstance as done in that commit only works for VE.Direct. Not for the /Devices on com.victronenergy.can0 for example: that service does not have a /DeviceInstance path; and also should not have one.

mpvader, 2019-06-21:
I've updated the __Devices__ section in the D-Bus documentation https://www.victronenergy.com/live/open_source:ccgx:d-bus. 

Also I added a small section to [the mqtt-rpc readme](https://github.com/victronenergy/mqtt-rpc/edit/master/README.md)

closed 2019-06-25 by mpvader