victronenergy/venus #476: systemcalc: store the relay state

mpvader, 2019-06-22:
Right now a reboots resets them to off.

Even if only switched off for a little while (because the SoC resets its gpio maybe/probably - I did not check), its better than now: reboot resets them to off.

When changed this way; perhaps the gui should control them via local settings rather than through systemcalc? Maybe, maybe not, but do make sure to reconsider the architecture when starting on this one.

izak, 2019-07-18:
There are 4 ways in which the relay can be used:

* Alarm relay, controlled by the GUI
* Generator start/stop, controlled by dbus-generator
* Tank pump, controlled by dbus-pump
* Manual

Upon a reboot, the first three will set the relay state correctly according to present circumstances, so this only affects the last manual case.

izak, 2019-08-19:
Released in v2.40~19.

closed 2019-08-19 by izak