victronenergy/venus #478: Mqtt-rpc doesn't always reconnect after some failures

wiebeytec, 2019-06-30:
Certain types of errors can cause the reconnect mechanism of mqtt-rpc/Paho to fail:

```
2019-06-30 20:46:01.099075500 Exception in thread Thread-1:                                                                                                                                                                                                                    
2019-06-30 20:46:01.099093500 Traceback (most recent call last):                                                                                                                                                                                                               
2019-06-30 20:46:01.099099500   File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner                                                                                                                                                                         
2019-06-30 20:46:01.099107500     self.run()                                                                                                                                                                                                                                   
2019-06-30 20:46:01.099111500   File "/usr/lib/python2.7/threading.py", line 754, in run                                                                                                                                                                                       
2019-06-30 20:46:01.099118500     self.__target(*self.__args, **self.__kwargs)                                                                                                                                                                                                 
2019-06-30 20:46:01.099125500   File "/opt/victronenergy/mqtt-rpc/paho/mqtt/client.py", line 2606, in _thread_main                                                                                                                                                             
2019-06-30 20:46:01.099283500     self.loop_forever(retry_first_connection=True)                                                                                                                                                                                               
2019-06-30 20:46:01.099293500   File "/opt/victronenergy/mqtt-rpc/paho/mqtt/client.py", line 1496, in loop_forever                                                                                                                                                             
2019-06-30 20:46:01.099301500     self.reconnect()                                                                                                                                                                                                                             
2019-06-30 20:46:01.099307500   File "/opt/victronenergy/mqtt-rpc/paho/mqtt/client.py", line 919, in reconnect                                                                                                                                                                 
2019-06-30 20:46:01.099314500     sock.do_handshake()                                                                                                                                                                                                                          
2019-06-30 20:46:01.099320500   File "/usr/lib/python2.7/ssl.py", line 848, in do_handshake
2019-06-30 20:46:01.099358500     match_hostname(self.getpeercert(), self.server_hostname)
2019-06-30 20:46:01.099365500   File "/usr/lib/python2.7/ssl.py", line 286, in match_hostname
2019-06-30 20:46:01.099372500     % (hostname, dnsnames[0]))
2019-06-30 20:46:01.099378500 CertificateError: hostname 'mqtt-rpc.victronenergy.com' doesn't match 'mqtt.victronenergy.com'
2019-06-30 20:46:01.100080500
```

The certificate error occurred, and it subsequentlly stopped trying to reconnect.

dbus-mqtt does do this properly, so look there for inspiration. Mqtt-rpc uses Paho's internal reconnect feature, whereas dbus-mqtt does that itself.

mpvader, 2019-11-04:
Izak will look at it.

Priority: its not a regression; this issue has been in the code for a long time. Up to Izak to decide to put it in v2.40; or the next one. Besides Izaks timeline this also depends on the risk.

@wiebeytec pls confirm; am I right that this is not a regression?

wiebeytec, 2019-11-04:
It's not a regression, but with the changes to the MQTT server I'm planning to do soon, I would feel more comfortable with a reconnect feature that **always** tries to reconnect.

mpvader, 2019-11-04:
Ok then its urgent. @izak

izak, 2019-11-11:
mqtt-rpc now connects to localhost (since Venus v2.40~29), so this is no longer an issue. Mosquitto handles the bridges to the mqtt servers.

closed 2019-11-11 by izak