victronenergy/venus #479: Remote VEConfigure: Add grid info readout support

vejpasop, 2019-07-02:
Upcoming Multi firmware will contain grid code information to be used by VEConfigure.
This data block provides information about the supported Grid code settings,

Todo: 
- [x] mk2vsc: Add grid code information readout support
- [ ] VRM Remote VEConfigure:  Add new mk2vsc status and error code
      - ```#define MK2VSC_ST_READ_GRID_INFO 24```
      - ```#define MK2VSC_ERR_GRID_INFO_READ 72```
- [ ] Release VEConfigure tooling supporting grid info

thlassche, 2019-07-03:
Related `mqtt-rpc` pull request, which adds support for the new states to VRM:

https://github.com/victronenergy/mqtt-rpc/pull/10

mpvader, 2019-07-11:
Added in v2.40~something, and will be backported to v2.33

closed 2019-07-11 by mpvader