victronenergy/venus #480: mqtt services and their settings

mpvader, 2019-07-02:
## Todo

- [ ] Can we indeed work with VictronConnect and MQTT over SSL?
- [ ] Review / finalise below design.

## Design

### 1. New settings and their functionality

We want to have two settings for enabling/disabling functionality:

1. MQTT on VRM   ( perhaps to be called VRM Two Way Communication)
2. MQTT on LAN

Functionality behind those two settings:

- If either is enabled, start both dbus-mqtt & mqtt-rpc & mosquitto
- If on VRM is enabled: enable the bridge config in mosquitto; otherwise disable it.
- If on LAN is enabled: open the inbound port (8883)

Then, there will be one more setting:

3. MQTT on LAN insecure - this can only be enabled if its secure variant is also enabled. It controls iptables inbound port 1883. This setting is for people that use MQTT themselves, ie. the tinkerers.

### 2. Migration:

- If the current Services -> MQTT setting is enabled then enable `MQTT on LAN` and `MQTT on VRM`; and also enable `MQTT on LAN insecure`, since that is how that works now and its used, for example, on certain externally managed ESS type installations.
- If the current VRM Two Way Communication setting is enabled, then enable only `MQTT on VRM`.

### 3. Security and passwords

Everything via VRM, is already secured. The connection between the local broker and the VRM broker is MQTT over SSL, using a Victron root CA certificate and pre-shared token. The connection between the client and the VRM-broker is also over SSL, and uses the VRM system for authentication.

We also want to secure local connections that use MQTT. The idea is to self generate a SSL certificate and private key pair on the Venus device at boot. And every new VictronConnect install that connects to that device, locally, will ask the user once to accept the certificate, showing its hash (aka finger print). On Remote Console, the same is visible, for cautious people to go and check before they accept it.

Also there will be a password on MQTT. But then preferably make a general password; instead of adding yet another one to the list (Remote Console & WiFi Access point). One option is to re-use the WiFi password for that on the Venus GX-es; in other words, promoting that to a general access password. Note that the CCGX does not have a factory installed password. Also note that the WiFi Password is currently not changeable on the Venus GX; which we can do; together with making the reset button work.

mpvader, 2019-07-10:
## Implemenation details for the MQTT on LAN (insecure) setting

- the gui needs a new setting, `Settings/System/MqttOnLanInsecure` named `MQTT on LAN (insecure)`. Implemented just like the local ssh port was done [1], but then controlling 1883 rather than port 22.
- add another migration to localsettings, just like [2]. But then it needs to check status of /Settings/Services/Mqtt; and when thats enabled, also enable MqttOnLanInsecure.

[1] https://github.com/victronenergy/gui/commit/972a978374e7ab13b56bb72d28a856e9490c30ee
[2] https://github.com/victronenergy/localsettings/commit/c23cbd49e4e73fddf720a9e284adbc114c4826f4

izak, 2019-09-05:
# Implementation details for MQTT bridges

From the original design above:

* If MQTT is disabled completely (all three settings are off), then all the MQTT-related processes are stopped (mqtt-rpc, dbus-mqtt, and mosquitto).
* If either MQTT on VRM or MQTT on LAN is enabled, all three services are started.
* The bridges are however only installed/enabled if MQTT on VRM is on. In the case where only MQTT on LAN (and its insecure sibling) is enabled, the bridges should be removed.

Presently the bridge configuration is managed by both mqtt-rpc and dbus-mqtt. Whichever process starts first will install the bridge. This should no longer be required in the new design.

## Proposal

* Remove bridge handling from dbus-mqtt
* Let mqtt-rpc manage both bridges
* mqtt-rpc will watch the settings and remove the bridge if required.

## Things to check:

* Does mosquitto handle removal/adding of bridges well, since we run an older mosquitto. See https://github.com/eclipse/mosquitto/issues/965

mpvader, 2019-09-10:
## 1 - Testing migration:

- only MQTT enabled -> mqtt on LAN, VRM Two Way and MQTT on LAN insecure all three enabled: PASS
- only VRM Two Way enabled -> only VRM Two Way enabled: PASS
- both enabled -> all three enabled: PASS

(after downgrading, remember to remove the newly made settings)

## 2 - Testing ip tables and bindings:

### 2.1 - Only MQTT on LAN enabled

Only VRM two-way enabled:

1883, 8883 and 9001 must all be disallowed.

```
root@beaglebone:~# iptables -nvL 
Chain INPUT (policy ACCEPT 45 packets, 9456 bytes)
 pkts bytes target     prot opt in     out     source               destination         
18053 3811K ACCEPT     all  --  lo     *       0.0.0.0/0            0.0.0.0/0           
 1234  415K new-conn   all  --  *      *       0.0.0.0/0            0.0.0.0/0            ctstate NEW

Chain FORWARD (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination         

Chain OUTPUT (policy ACCEPT 290 packets, 135K bytes)
 pkts bytes target     prot opt in     out     source               destination         

Chain new-conn (1 references)
 pkts bytes target     prot opt in     out     source               destination         
    2   128 ACCEPT     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:22
    0     0 REJECT     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:1883 reject-with icmp-port-unreachable
    0     0 REJECT     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:9001 reject-with icmp-port-unreachable
    0     0 REJECT     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:8883 reject-with icmp-port-unreachable
```

PASS

### 2.2 - Enabling MQTT on LAN

Allow 8883:

```
root@beaglebone:~# iptables -nvL 
Chain INPUT (policy ACCEPT 45 packets, 8360 bytes)
 pkts bytes target     prot opt in     out     source               destination         
26784 4763K ACCEPT     all  --  lo     *       0.0.0.0/0            0.0.0.0/0           
 1840  596K new-conn   all  --  *      *       0.0.0.0/0            0.0.0.0/0            ctstate NEW

Chain FORWARD (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination         

Chain OUTPUT (policy ACCEPT 123 packets, 173K bytes)
 pkts bytes target     prot opt in     out     source               destination         

Chain new-conn (1 references)
 pkts bytes target     prot opt in     out     source               destination         
    2   128 ACCEPT     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:22
    0     0 REJECT     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:1883 reject-with icmp-port-unreachable
    0     0 REJECT     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:9001 reject-with icmp-port-unreachable
    0     0 ACCEPT     tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            tcp dpt:8883
```

Allows 8883: PASS.

### 2.3 - Enable MQTT on LAN (insecure)

```
oot@beaglebone:~# iptables -nvL 
[... 55 lines cut ...]
PASS for all switch combinations.

### 3.2 - mqtt-rpc works on VRM when VRM two-way communication is enabled.

PASS

### 3.3 - VRM two-way communication

PASS (both for enabled and disabled).

### 3.4 - dbus-mqtt.. ??

skipped; I don't know how to easily test this.

### 3.5 - Start without settings aka default settings:

```
svc -d /service/localsettings
rm /data/conf/settings.xml
rm -rf /data/conf/mosquitto.d
rm /data/conf/mqtt_password.txt
```

- generation of mqtt bridge auth: PASS (triggers the vrm portal to ask for a reset which is OK)
- all mqtt services default off:  PASS