victronenergy/venus #482: ESS: No ESS Assistant found.

vejpasop, 2019-07-03:
Venus V2.40~8

After Multi firmware update to 2608459 and installation of ESS assistant (137) 

GUI shows:
[image]

While an ESS assistant is installed

mk2-dbus info.
[image]

jhofstee, 2019-07-03:
@mpvader Systemcalc is crashing on this line https://github.com/victronenergy/dbus-systemcalc-py/blob/master/dbus_systemcalc.py#L490. 

```
@400000005d1c937614ba8a9c Traceback (most recent call last):
@400000005d1c937614c3f0dc   File "/opt/victronenergy/dbus-systemcalc-py/dbus_systemcalc.py", line 921, in <module>
@400000005d1c937614e02784     systemcalc = DbusSystemCalc()
@400000005d1c937614e63264   File "/opt/victronenergy/dbus-systemcalc-py/dbus_systemcalc.py", line 275, in __init__
@400000005d1c937614f063c4     self._updatevalues()
@400000005d1c937614f549ac   File "/opt/victronenergy/dbus-systemcalc-py/dbus_systemcalc.py", line 490, in _updatevalues
@400000005d1c93761502b344     newvalues['/Dc/Pv/Power'] = v * (i + l)
@400000005d1c937615093354 TypeError: unsupported operand type(s) for +: 'float' and 'NoneType'
```

mpvader, 2019-07-03:
Closing this one; #483 is the real bug.

closed 2019-07-03 by mpvader