victronenergy/venus #484: sshd not running

mpvader, 2019-07-04:
we had a canvu on which sshd wasn’t working. Due to the fingerprint key pair (is that what they’re called?) being empty files. Not the first time that we’ve seen empty files somewhere.

Sshd doesn’t start in that case.

The fix will likely be to remove empty files before starting sshd; or similar.

jhofstee, 2019-07-05:
please close the issue when it is actually included...

jhofstee, 2019-07-05:
ah right, my bad. When adding it to the todo, please add that it closes an issue once it is included. As in closes https://github.com/victronenergy/venus/issues/484. If the venus maintainer copy paste that properly if will close when included.

closed 2019-07-05 by jhofstee

jhofstee, 2019-07-05:
For completeness, with the full url of the issue, github will then be able to find across different accounts / repositories.