victronenergy/venus #486: CCGX: GUI: hangs when left button 'out' from ESS menu

DanielBoekel, 2019-07-11:
CCGX: then going to quickly from ESS menu (after changing mode), using left button to exit, once in a while interface hangs. 

Also had this (GUI lockup) while accessing a Venus device in a level3 case, case 2333

mpvader, 2019-11-04:
Hey @DanielBoekel have you been able to reproduce it? Jeroen (the developer) tried too; but couldn't. Thats why the issue is on hold. He has an idea what it could be (focus issue); but still needs to reproduce before moving on.

DanielBoekel, 2019-11-04:
Hi @mpvader 
I've experienced a couple of times more, I can try to do this on purpose and leave the CCGX in this state...

jhofstee, 2020-06-25:
almost one year later, but I can finally reproduce this, it is related to https://github.com/victronenergy/venus/issues/623. It happens when a new page transition is started during a page transition.

closed 2020-06-30 by jhofstee