victronenergy/venus #487: dbus-mqtt: System Alarms from VEBus are not pushed on MQTT

Wholistic, 2019-07-12:
'Warnings' are pushed, but the more critical 'Alarms' are not.
If the Multi shuts down due to overload, this is not visible.

Next level systems like influxdb via MQTT miss this important info which is kind of essential for troubleshooting systems with shut downs. 

This information is revealed on VRM, but 1 minute sample rates can be too slow to narrow down momentary overloads:

[image]

Influxdb via MQTT has the moment by moment loads, but makes it difficult to find the right place to be looking without the Alarms:

[image]

izak, 2019-07-12:
Initial guess (to be confirmed): The rate limiter likely kills the alarm state by delaying it, and then it is removed from the broker when the service goes away. Possible solution: Flush everything when a service goes away.

mpvader, 2019-07-12:
@Wholistic wrt vrm:

any overload alarm (= shutdown) longer than a few seconds will be registered.

Despite the 1 minute minimum interval. Events like that trigger and immediate transmission

izak, 2019-07-18:
@Wholistic I tested my theory and sadly (or perhaps not?) I was wrong. The alarms make it onto mqtt quite succcessfully. By subscribing to the events on the test unit I got the whole range, both low battery and overload transitioning to Alarm and back:

```
N/d0ff500097c0/vebus/257/Alarms/LowBattery {"value": 1}
N/d0ff500097c0/vebus/257/Alarms/Overload {"value": 1}
N/d0ff500097c0/vebus/257/Alarms/L1/LowBattery {"value": 1}
N/d0ff500097c0/vebus/257/Alarms/L1/Overload {"value": 1}
N/d0ff500097c0/vebus/257/Alarms/Overload {"value": 2}
N/d0ff500097c0/vebus/257/Alarms/L1/Overload {"value": 2}
N/d0ff500097c0/vebus/257/Alarms/Overload {"value": 1}
N/d0ff500097c0/vebus/257/Alarms/L1/Overload {"value": 1}
N/d0ff500097c0/vebus/257/Alarms/LowBattery {"value": 0}
N/d0ff500097c0/vebus/257/Alarms/Overload {"value": 0}
N/d0ff500097c0/vebus/257/Alarms/L1/LowBattery {"value": 0}
N/d0ff500097c0/vebus/257/Alarms/L1/Overload {"value": 0}
```

This was using a 12V 3kVA Multiplus and just doing a dead short on the output.

mpvader, 2019-10-14:
This must be some other issue, perhaps how it was queried or used in Grafana, or specific to this 
 site, I don't know. closing for now.

I've seen it work myself, in Grafana, too, a while ago.

closed 2019-10-14 by mpvader