victronenergy/venus #489: GUI: DBus SetValue not called when trying to set the same value again

vejpasop, 2019-07-16:
Issue occasionally appears during Multi off/on switching through the GUI and can be easily reproduced with a slightly modified mk2-dbus and a Quattro. (On this type Multi the main switch always switches off the MCU power)

Affected DBus path ```com.victronenergy.vebus.tty##/Mode```

**Mk2-dbus modification:**
Removed ```veItemOwnerSet``` call from the SetValue function.
With this modification in place the "Mode" path will only signal a PropertiesChanged when the requested Multi on/off action was successful.

**Steps to reproduce the issue:**

1.  Switch Quattro  "off" through GUI.
2. Switch the Quattro main switch off. (disconnects MCU power)
3. Switch Quattro "on" through GUI
	a. SetValue is called with "on"
	b. GUI switch setting shows "off"
	c. Multi does not reply since it is switched off by its main switch. So we will not get a ```PropertiesChanged```
4. Switch Quattro "on" through GUI. 
	a. No SetValue any more.  
		
So even if we switch the Quattro main switch back on we will not be able to switch is back on through the GUI. Quattro remains "off" because it remembered it was switched off though the GUI before.

Writing an "on" value to the ```/mode``` path through dbus-spy works properly.

jhofstee, 2019-08-19:
Indeed, the SetValue should be answered with one of these:
- error
- or a property change, as in veItemOwnerSet
- (or a NameOwnerChanged if the whole thing crashed, but the dbus deamon will do that)

If dbus-mk2 doesn't do any of that, it is a bug of dbus-mk2. I will discuss this later with you.

mpvader, 2019-11-04:
Jeroen will make something for this

jhofstee, 2019-11-11:
https://github.com/victronenergy/velib/tree/WIP_force_changed should be able to fix this

vejpasop, 2019-11-12:
Applied in mk2-dbus v2.99

closed 2019-11-12 by jhofstee