victronenergy/venus #491: BMV-712 stopped reporting

drolfe, 2019-07-19:
Hi,

I've been using v2.33 on a pie3 for around 6 months without issues, my MPPT150/35 and BMV-721 have been working fine, I did some works recently and installed a new inverted (non victron) but since getting everything connected back up the BMV-712 has stopped showing in the device list for VRM portal and via the venus console

It's still working via bluetooth and reporting batt amps, volts and AH usage as per normal

Just wondering how I can T-Shoot this further ?

I did try to sway the USB cables between the BMV and MPPT, this didn't fix the issue and the MPPT reporting was still working on the original BMV USB cable

Any help to debug would be great

Regards, Daniel

drolfe, 2019-07-22:
Actually I just checked VRM today and it's started reporting, however it seems to be dropping in and out as per the attached. 
Any ideas what could be causing this ?

[image]

drolfe, 2019-07-23:
hmm very strange looks like it's stopped reporting again.
[image]

mpvader, 2019-07-23:
Hi @drolfe, this is most likely a cable/connection issue. You seem to have ruled out the cable (though I dont know what cable you are using but you can never be sure) so that leaves the port on the BMV.

I’ll close this issue, its not something for r&d.

Please see www.victronenergy.com/support in case you need any help. Either a dealer (though that might be a bit difficult with a Pie); and otherwise the Modifications space on community or the dev mailing list.

closed 2019-07-23 by mpvader

drolfe, 2019-08-12:
Ok I've done some more debug on this

When I connect the Ve.direct USB dmesg shows the normal things

`[  972.941914] usb 1-1.2: new full-speed USB device number 7 using dwc_otg
[  973.103615] usb 1-1.2: New USB device found, idVendor=0403, idProduct=6015
[  973.103627] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  973.103634] usb 1-1.2: Product: VE Direct cable
[  973.103642] usb 1-1.2: Manufacturer: VictronEnergy BV
[  973.103649] usb 1-1.2: SerialNumber: VE2NHXGJ
[  973.112960] ftdi_sio 1-1.2:1.0: FTDI USB Serial Device converter detected
[  973.113158] usb 1-1.2: Detected FT-X
[  973.114362] usb 1-1.2: FTDI USB Serial Device converter now attached to ttyUSB0`

drolfe, 2019-08-12:
However I keep a checksome error on the ve.directUSB0 logs, likely do the the below TAG issue:

[  972.941914] usb 1-1.2: new full-speed USB device number 7 using dwc_otg
[  973.103615] usb 1-1.2: New USB device found, idVendor=0403, idProduct=6015
[  973.103627] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  973.103634] usb 1-1.2: Product: VE Direct cable
[  973.103642] usb 1-1.2: Manufacturer: VictronEnergy BV
[  973.103649] usb 1-1.2: SerialNumber: VE2NHXGJ
[  973.112960] ftdi_sio 1-1.2:1.0: FTDI USB Serial Device converter detected
[  973.113158] usb 1-1.2: Detected FT-X
[  973.114362] usb 1-1.2: FTDI USB Serial Device converter now attached to ttyUSB0

see the full error message below:

@400000005d50cf190828dc0c 251.526 WRN.VE.Frame: Invalid checksum: 60
@400000005d50cf301a53a59c 274.831 INF.VE.Text: Tag not found: I%�JR$�ʕ56=56
@400000005d50cf301a53c4dc 274.831 INF.VE.Text: CE=0
@400000005d50cf301a53ccac 274.831 INF.VE.Text: SOC=1000
@400000005d50cf301b39f68c 274.846 INF.VE.Text: TTG=-1
@400000005d50cf301b3a062c 274.846 INF.VE.Text: ALARM=OFF
@400000005d50cf301b3a11e4 274.846 INF.VE.Text: RELAY=OFF
@400000005d50cf301c339cb4 274.862 INF.VE.Text: AR=0
@400000005d50cf301c33a86c 274.862 INF.VE.Text: BMV=712 SMART
@400000005d50cf301c33b424 274.863 INF.VE.Text: FW=0401
@400000005d50cf301d2c50ac 274.879 WRN.VE.Frame: Invalid checksum: 41
@400000005d50cf6621c9ba14 328.956 INF.VE.Text: Tag not found: I%�JR$�ʕ0=0
@400000005d50cf6621cb96a4 328.956 INF.VE.Text: H3=-65331
@400000005d50cf6622af8a1c 328.971 INF.VE.Text: H4=4
@400000005d50cf6622b00b04 328.971 INF.VE.Text: H5=0
@400000005d50cf6622b016bc 328.971 INF.VE.Text: H6=-5508856
@400000005d50cf6623a9536c 328.988 INF.VE.Text: H7=5954
@400000005d50cf6623a9d06c 328.988 INF.VE.Text: H8=20478
@400000005d50cf6623a9d83c 328.988 INF.VE.Text: H9=0
@400000005d50cf6623aa4d6c 328.988 INF.VE.Text: H10=15
@400000005d50cf6624a33fe4 329.004 INF.VE.Text: H11=0
@400000005d50cf6624a3bce4 329.004 INF.VE.Text: H12=0
@400000005d50cf6624a3c89c 329.004 INF.VE.Text: H15=27
@400000005d50cf6624a439e4 329.004 INF.VE.Text: H16=21022
@400000005d50cf66259d18d4 329.020 INF.VE.Text: H17=6715
@400000005d50cf66259d9da4 329.020 INF.VE.Text: H18=12082
@400000005d50cf66259da95c 329.020 WRN.VE.Frame: Invalid checksum: 87
@400000005d50cfaf058b6d0c 401.482 INF.VE.Text: Tag not found: I%�JR$�ʕ0=0
@400000005d50cfaf058b8864 401.482 INF.VE.Text: H3=-65331
@400000005d50cfaf06749874 401.498 INF.VE.Text: H4=4
@400000005d50cfaf0674b7b4 401.498 INF.VE.Text: H5=0
@400000005d50cfaf0674bf84 401.498 INF.VE.Text: H6=-5508856
@400000005d50cfaf076c2b5c 401.514 INF.VE.Text: H7=5954
@400000005d50cfaf076c3714 401.514 INF.VE.Text: H8=20478
@400000005d50cfaf076c42cc 401.514 INF.VE.Text: H9=0
@400000005d50cfaf076c4a9c 401.514 INF.VE.Text: H10=15
@400000005d50cfaf0865c5cc 401.530 INF.VE.Text: H11=0
@400000005d50cfaf0865d56c 401.530 INF.VE.Text: H12=0
@400000005d50cfaf0865dd3c 401.530 INF.VE.Text: H15=27
@400000005d50cfaf0865e50c 401.530 INF.VE.Text: H16=21022
@400000005d50cfaf094cab2c 401.545 INF.VE.Text: H17=6715
@400000005d50cfaf094cb6e4 401.545 INF.VE.Text: H18=12082
@400000005d50cfaf094cc29c 401.545 WRN.VE.Frame: Invalid checksum: 06
@400000005d50cfe62b617224 457.117 INF.VE.Text: Tag not found: I%�JR$�ʕ0=0
@400000005d50cfe62b631bec 457.117 INF.VE.Text: H3=-65331
@400000005d50cfe62c597a3c 457.133 INF.VE.Text: H4=4
@400000005d50cfe62c5a35bc 457.133 INF.VE.Text: H5=0
@400000005d50cfe62c5a4174 457.134 INF.VE.Text: H6=-5508856
@400000005d50cfe62d53438c 457.150 INF.VE.Text: H7=5954
@400000005d50cfe62d53b8bc 457.150 INF.VE.Text: H8=20478
@400000005d50cfe62d542234 457.150 INF.VE.Text: H9=0
@400000005d50cfe62d5487c4 457.150 INF.VE.Text: H10=15
@400000005d50cfe62e3b134c 457.165 INF.VE.Text: H11=0
@400000005d50cfe62e3bbb44 457.165 INF.VE.Text: H12=0
@400000005d50cfe62e3bc6fc 457.165 INF.VE.Text: H15=27
@400000005d50cfe62e3c3844 457.165 INF.VE.Text: H16=21022
@400000005d50cfe62f33e684 457.181 INF.VE.Text: H17=6715
@400000005d50cfe62f345f9c 457.181 INF.VE.Text: H18=12082
@400000005d50cfe62f34c52c 457.181 WRN.VE.Frame: Invalid checksum: D8