victronenergy/venus #494: VE.Bus: Add readout of network quality counters

vejpasop, 2019-07-26:
VE.Bus network quality counters can be readout since Multi firmware version 460. 

These counters can be useful for debugging VE.Bus communication problems

mk2-dbus v2.94 publishes the counters on the dbus through ```Devices/#/ExtendStatus/VeBusNetworkQualityCounter```

The counters are updated upon initialization of mk2-dbus, after a VE.Bus error and approximately every 5 minutes.

Todo:

- [ ] Add to VRM logger
- [ ] Add to GUI

mpvader, 2019-10-07:
Hi Jos,

screenshot please. And when do they reset to 0? And these are the counters in the devices themselves, right? Ie not the counters from the MK3 in the GX Device?

And in what event(s) does the counter increment?

vejpasop, 2019-10-08:
Hi @matthijs,

I added the screenshot and story to the description above. 
Indeed these are the counters in the devices themselves, and not in the MK3.

We did not implement these counters in the MK3 firmware (yet)

closed 2019-10-14 by mpvader