victronenergy/venus #499: Venus OS MFD integration on Simrad makes the plotter unresponsive

k-ljung, 2019-08-08:
I'm running the latest Venus OS on my Venus GX connected to my local lan on my boat, I have one Simrad NSS12 Evo 2 and one Simrad NSS7 Evo 3 conntected to the same lan.

The Victron icon is displayed on both plotters and I can access the Victron UI.

But the Victron icon "flickers" a bit on both plotters like every 10:s second, like it re-connect to the Venus, the Venus integration also makes the plotters a bit unresponsive when navigating on the home page and in settings page, sometimes I press an icon on homepage and the sound of pressing the screen is sounded but nothing happens, I then press again and the plotter loads the selected feature.

If I disconnect the Venus from my lan and when the Victron icon disappear from the plotters home page, the plotters works as expected and are much more responsive when navigating in menus, settings etc.

More users experience the same, https://community.victronenergy.com/questions/20970/venus-os-mfd-integration-on-simrad-makes-the-plott.html?childToView=20982#comment-20982

jhofstee, 2019-08-19:
fyi: this is being investigated together with Navico

mpvader, 2019-11-04:
@martinbosma did you speak to Navico about this? Any status? Has it been reproduced?

martinbosma, 2019-11-05:
I have asked them already in August, but haven't heard from them since. I will ping them again.
I cannot reproduce, everything is working smoothly on my setup.

martinbosma, 2019-11-13:
I was able to reproduce the issue on my desk. When the MFD is connected directly to the GX device, it does not occur, probably because both devices use a link local address. However, when I connect both devices via a router (or other device that has an DHCP server), the GX devices announces itself every 10 seconds on two IP adresses. 

Wireshark dump:
```
No.     Time           Source                Destination           Protocol Length Info
    264 8.612027       192.168.10.101        239.2.1.1             UDP      442    52534 → 2053 Len=400
    268 8.649069       169.254.10.134        239.2.1.1             UDP      442    60702 → 2053 Len=400
    450 18.693925      192.168.10.101        239.2.1.1             UDP      442    33444 → 2053 Len=400
    451 18.729548      169.254.10.134        239.2.1.1             UDP      442    47019 → 2053 Len=400
```

k-ljung, 2019-11-13:
Make sense, I have connected my Evo2 and Evo3 through a router.

martinbosma, 2020-01-13:
According to Navico, sending conflicting announcements have been verified to cause performance problems - especially on the home page.

Sending the announcement on the link local address only should solve the issue. @mansr Can you fix this?

closed 2020-01-16 by jhofstee