victronenergy/venus #503: Maxi GX not requesting device information from all BMS's by CAN-bus

wilcop85, 2019-08-24:
Build test setup with following components:
- 3 Quattro's configure 3 phase ESS
- 15 MPPT's connected by VE-Direct
- 3 Lynx ION BMS with DI: 2, 4 and 6.
[image]

When Maxi GX is reboot, not always all the BMS's are detected. In the attachment excel file there is a CANdump of the reboot. So far I can see is that all the BMS's respond on the ACL request but the Maxi GX is only asking 2 BMS's for device information.

In the Excel file there is also a VUP request with simular response.

Note two tabs -> [canDumps.xlsx](https://github.com/victronenergy/venus/files/3537114/canDumps.xlsx)

https://vrm.victronenergy.com/installation/42381/dashboard

mpvader, 2019-08-24:
What Venus OS versions did you test & see this behaviour on @wilcop85 ?

Please make sure to also test with latest version now in testing (v2.40~22).

mpvader, 2019-08-24:
as discussed elsewhere, you havent tested that yet. To install that version, use `/opt/victronenergy/swupdate/set-feed.sh testing`

More details here: https://github.com/victronenergy/venus/wiki/release-cycle

wilcop85, 2019-08-24:
Updated to v2.40~22
No Lynx Ion BMS displayed anymore. With VUP BMS's are found but in dbus-spy they are not present.

[image]
[image]

jhofstee, 2019-08-29:
@wilcop85 it is not the Venus version, just installed v2.40~22 again and it does find the (some) BMSes. It is not vup versus dbus-vecan either, look at your screenshot above, the first vup run also misses a product id. The problem seems to be that can bus messages are dropped somewhere. Can you update the octo to testing as well, connect it to the same CAN-bus and provide the VRM url of it, so it can be checked remotely if the messages are there or really missing.

fyi: you can run dup -d or dup -dd to dump the messages it receives.

wilcop85, 2019-08-30:
@jhofstee Did you have a look a the excel file? The Maxi is not sending a request for more device information.

I had v2.40~22 installed 6 days ago, first time it booted it didn't  find any BMS's. Rebooting it a view more times it did find some BMS's but not all.

Meanwhile I did some more tests with the system and therefor I used V2.33, thats the reason you dint see it. I also connected two more BMS's, total now are 5.

I just rebooted the MaxiGX and it only finds 3 of the 5 BMS's. When I perform a VUP it finds one more.

I will try with the octo later on this morning. Maybe it's handy to continue on slack?

jhofstee, 2019-08-30:
I don't understand what you mean. As I told you both vup and the app have a problem on all versions of venus. So what do you mean with "thats the reason you dint see it"?

jhofstee, 2019-08-30:
just have a look which version you are running now ;)

wilcop85, 2019-08-30:
The octo with version 2.33 seems to be working better. rebooded it some times now and connected 10 mppts rebooded again and it is finding the bms's

wilcop85, 2019-08-30:
@jhofstee I have reconnected everything to the Maxi again so you can do your thing ;) It detected 4 of the 5 Lynx Ion BMS's

wilcop85, 2019-08-30:
@jhofstee 

CCGX dbus-spy
[image]

Maxi dbus-spi
[image]

mpvader, 2019-08-31:
Fyi @jhofstee: Wilco was working on a test with 25 ve.direct mppts with real pv panels and parallel bms-es, on a maxi gx: to make sure it works as there are now a few systems like that being installed. I told Wilco to implement the one minute ACL send to sidestep this nanopi/maxigx/slcan canbus problem, so he can continue with his test.

And in the meantime we can work on the canbus problem.

mpvader, 2019-08-31:
Also, I explained Wilco that we decoupled the canbus instance from the vrm instance / instance on /DeviceInstance on VRM. And to also sidestep that for now, he continues first on v2.33.

mpvader, 2019-08-31:
I wrote:

> Fyi @jhofstee: Wilco was working on a test with 25 ve.direct mppts with real pv panels and parallel bms-es, on a maxi gx: to make sure it works as there are now a few systems like that being [...]

Too bad: that didn’t work. The BMS-es already send the ACL message every minute. And vecan-dbus on the maxi just doesnt respond (or it does, but then the messages it wants to send are dropped: kvaser and dump on the Maxi GX don’t show a reaction to those ACLs sent by the BMSes). Running vup does make the devices show up; I saw it myself just now. v2.33. And like mentioned above also vup is having issues. Various bits of info are missing and show up as an `x`. Which ones do differs per run.

wilcop85, 2019-09-01:
> I wrote:
> 
> > Fyi @jhofstee: Wilco was working on a test with 25 ve.direct mppts with real pv panels and parallel bms-es, on a maxi gx: to make sure it works as there are now a few systems like that being installed. I told Wilco to implement the one minute ACL send to sidestep this nanopi/maxigx/slcan canbus problem, so he can continue with his test.
> 
> Too bad: that didn’t work. The BMS-es already send the ACL message every minute. And vecan-dbus on the maxi just doesnt respond (or it does, but then the messages it wants to send are dropped: a log made with a kvaser doesn’t show a reaction to those ACLs sent by the BMSes). Running vup on the maxi does make the devices show up; I saw it myself just now. v2.33. And like mentioned earlier in this issue, also vup is having issues. Various bits of info are missing and show up as an `x`. Which ones do differs per run.

Yes this is correct. The maxi thinks it is sending CAN-messages but they dont appear on the CAN-bus. With a "vup -dd" you can see that the maxi is requesting for more device information. But in the log made with extern kvaser not all this request are actually on the CAN-bus. 0x51, 0x52 and 0x53 are missing. The once missing are random.
[image]
[image]

[image]

mpvader, 2019-09-01:
I built & loaded a new swu file for that nanopi including kvaser driver; and made it work.

So, there is a kvaser leaflight usb can interface connected to the external usb; and the 'internal' slcan connected both to the same canbus; having 6 BMS-es.

Result: on the kvaser; all is stable. On the slcan, packets are lost. See vup result.

```
root@nanopi:~# ip link show can0
5: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UNKNOWN mode DEFAULT group default qlen 10
    link/can 

root@nanopi:~# ip link show can1
7: can1: <NOARP,UP,LOWER_UP> mtu 16 qdisc pfifo_fast state UNKNOWN mode DEFAULT group default qlen 10
    link/can 

root@nanopi:~# vup -c socketcan:can0
Querying devices
| NAD  | man | serial              | boot   | bmode| vup| vup#| id   | description                      | version  |
| 0x50 | 358 |                     | 1.04   | -    | +  | 0   | A392 | Lynx Ion BMS 400A                | 1.19     |
| 0x51 | 358 |                     | 1.04   | -    | +  | 0   | A392 | Lynx Ion BMS 400A                | 1.19     |
| 0x52 | 358 |                     | 1.04   | -    | +  | 0   | A392 | Lynx Ion BMS 400A                | 1.19     |
| 0x53 | 358 |                     | 1.04   | -    | +  | 0   | A392 | Lynx Ion BMS 400A                | 1.19     |
| 0x54 | 358 |                     | 1.04   | -    | +  | 0   | A392 | Lynx Ion BMS 400A                | 1.19     |
| 0x55 | 358 |                     | 1.04   | -    | +  | 0   | A392 | Lynx Ion BMS 400A                | 1.19     |
| 0xE0 | 358 |                     | x      | x    | x  | x   | x    | x                                | x        |
| 0xF1 | 358 | serial#             | x      | x    | x  | x   | x    | x                                | x        |
root@nanopi:~# vup -c socketcan:can1
Querying devices
| NAD  | man | serial              | boot   | bmode| vup| vup#| id   | description                      | version  |
| 0x50 | 358 |                     | x      | x    | x  | x   | x    | x                                | x        |
| 0x51 | 358 |                     | 1.04   | -    | +  | 0   | A392 | Lynx Ion BMS 400A                | 1.19     |
| 0x52 | 358 |                     | 1.04   | -    | +  | 0   | A392 | Lynx Ion BMS 400A                | 1.19     |
| 0x53 | 358 |                     | x      | x    | x  | x   | x    | x                                | x        |
| 0x54 | 358 |                     | x      | x    | x  | x   | x    | x                                | x        |
| 0x55 | 358 |                     | 1.04   | -    | +  | 0   | A392 | Lynx Ion BMS 400A                | 1.19     |
| 0xE0 | 358 | serial#             | x      | x    | x  | x   | x    | x                                | x        |
| 0xF1 | 358 | serial#             | x      | x    | x  | x   | x    | x                                | x        |
root@nanopi:~# 
```

mpvader, 2019-09-01:
In a quest to see if tx packets are dropped somewhere, and where, I ran vup -dd on both interfaces, and checked the increase in the tx counters provided by ifconfig. For both interfaces the same number showed: 17 packets are sent by vup. No difference.

mpvader, 2019-09-09:
This resulted in the conclusion that the VE.Can port on the Maxi GX is not suitable for VE.Can, and -with its current hardware- never will be.

New hardware is being worked on.

And in the meantime, Venus OS will be changed to fix this canbus port to BMS-Can only, 500kbps. see #507

mpvader, 2019-09-09:
And many thanks to @wilcop85 for finding, raising and helping to pin this issue down!

closed 2019-09-09 by mpvader