victronenergy/venus #505: VE.Bus: Error 11 split-up

vejpasop, 2019-08-30:
In upcoming VE.Bus firmware (version 464??) VE.Bus error 11 will split-up in to two different error codes.

- Error 8 = BFRelayTestError
- Error 11 = GNDRelayTestError

The error reason from both errors can still be read from the ```Devices/#/ExtendStatus/GridRelayReport/...``` items.

Note:  The current Venus release only updates the ```GridRelayReport``` only upon initialization and after an error 11.  From Venus OS 2. 40~17 (mk2-dbus v2.94) onward the ```GridRelayReport``` always gets updated after any vebus error and every 5 minutes.

Todo:
- [ ] mk2-dbus add Error 8 definition and description
- [ ] GUI Add error 8 description and update "VE.Bus Error 11 report" text 
- [ ] update https://www.victronenergy.com/live/ve.bus:ve.bus_error_codes

mpvader, 2019-09-10:
https://www.victronenergy.com/live/ve.bus:ve.bus_error_codes#error_8_-_ground_relay_test_failed

mpvader, 2019-11-04:
I (a) defined error 8 in the database and (b) added lots of others too; so they do show up in logs even if not named yet.

[image]

closed 2019-11-05 by jhofstee