victronenergy/venus #511: Force good settings for BYD batteries

izak, 2019-09-11:
When a BYD battery is detected, apply the following settings:

DVCC = on
SVS = off
STS = off

Extend the possible values for these settings, so that:

0 = off
1 = on
2 = force off
3 = force on

Extend the GUI display to display the correct value and make it read-only when the value is forced.

mpvader, 2019-09-13:
MG electronics: DVCC force on, SVS force off, STS force off

izak, 2019-09-16:
There are some corner cases here.

We want to automatically force the setting (and lock/hide it in the GUI) if we know a good set of values for it.

* What if there is more than one battery in the system of different types? Answer: Only do this for the selected battery monitor.
* And if the battery monitor changes? Answer: Then unlock the setting and leave it where it was.

Therefore the locking/unlocking code fires on: 1) A new battery is detected, 2) it's a battery we have known-good settings for, 3) this battery is the battery monitor (either auto- or manually selected).

For this to work well, a way is needed for the delegates to see when the battery monitor changes. This does not exist yet.

izak, 2019-09-17:
Some GUI decisions. When DVCC is locked on, remove the firmware notice. It is pointless to warn the user over something he has no control over.

[image]

Forced settings looks like this:

[image]

mpvader, 2019-09-17:
This links to another issue we have open.: raise a warning about too old firmware; there is an issue for that.

I hadn’t thought about this one; it sounds like now the moment has come that we need to make that warning. Do it in systemcalc so that wanring is available both in gui & on vrm?

mpvader, 2019-09-26:
@izak you wrote:

> For this to work well, a way is needed for the delegates to see when the battery monitor changes. This does not exist yet.

Why not evaluate this on each iteration in the main loop?

Then you’ll be sure to never miss an event.

Check active main battery type against the actual values of the dvcc & svs & sts settings.

izak, 2019-09-26:
I found a reliable spot to hook this so that we don't have to do the entire (unnecessary) calculation on every loop.

Systemcalc already calculates the active battery monitor every time and updates dbus only if it changes:

https://github.com/victronenergy/dbus-systemcalc-py/blob/master/dbus_systemcalc.py#L337

So I extended this to delegates.

The forced settings are applied at startup or when the battery is added too.

https://github.com/victronenergy/dbus-systemcalc-py/blob/master/dbus_systemcalc.py#L337

mpvader, 2019-09-26:
👍

closed 2019-09-30 by jhofstee