victronenergy/venus #517: Improve GX GSM Manual

mpvader, 2019-10-03:
While helping someone with a GX GSM problem, I improved the manual. The most important chapter is chapter 5, Troubleshooting. Pls review how it is now. If there is any other intel available for customers, we should add it; it will be appreciated.

@mansr / @jepefe pls review; & make changes if you see any, or let me know; then I'll change it.

https://www.victronenergy.com/live/venus-os:gx-gsm

btw, the badly rendered table with the icons (its clipped at the right), is being fixed.

mkrenzke, 2019-10-03:
After fix of numbering "Trouble shooting" is point 6 now.

Also I've added a list of SIM card states to the "Status" chapter and did some improvements in formatting.

mpvader, 2019-11-04:
good enough

closed 2019-11-04 by mpvader