victronenergy/venus #518: Generator start/stop: Get readings from systemcalc

jepefe, 2019-10-07:
Currently the dbus-generator service gets the system configured battery service instance from systemcalc and then it reads the voltage, current and SOC values directly from the service.

As systemcalc offers in dbus the DC and AC readings from the respective services we can use it as an abstraction layer and simplify the dbus-generator code dropping all the underlying handling due to the direct connection to the battery/inverter services.

Systemcalc provides calculated values that can be used in systems without a battery monitor to properly evaluate the start/stop conditions.

This user requested us to make the generator start/stop functionality be aware of PV power:
https://community.victronenergy.com/questions/17752/trigger-generator-change-to-acin-based-on-battery.html

Proposal:
- Switch to use systemcalc but keep getting readings directly from the selected battery service for those systems not using same battery service than systemcalc.
- Don't use calculated values for the existing conditions. Do not change the current behaviour of the existing conditions, users are used to them and it is a bit risky. 
- Add a new condition or a set of "PV coupled" conditions, these conditions will optimise the usage of PV power by getting the calculated values from systemcalc.

Of course all this needs to be done in a simple way to not confuse existing users.

mpvader, 2019-10-07:
Hi Jesus, I'm not (yet?) worried about using the calculated values for existing conditions. By using them always:

a) nothing will change for systems with a real battery monitor being used in Settings -> System Setup. 
b) for systems with Inverter/charger(s) & Solar chargers, finally it will use the battery current, rather than just the DC current as reported by in the inverter/charger system.

jepefe, 2019-10-08:
> a) nothing will change for systems with a real battery monitor being used in Settings -> System Setup.

Right, this is the best case nothing will change here

> b) for systems with Inverter/charger(s) & Solar chargers, finally it will use the battery current, rather than just the DC current as reported by in the inverter/charger system.

Yes, so after the changes will work based on the actual(calculated) battery current. 
My only concern are users that already have set the start/stop conditions based on the VE.Bus DC values, they probably must switch to use AC values instead. Changing it will probably make their system stop working as it used to do in previous versions.

What do you think, should we worry about this or not? If not, it will be easier as we don't need to add extra conditions.

mpvader, 2019-10-08:
ok understood.

but indeed: don't worry about that. No need to add extra conditions; better not even, and take the pain now instead of for ever onwards.

mpvader, 2019-11-04:
Jesus, whats your plan on this timewise?

jepefe, 2019-11-04:
I've done the changes already I need to create and modify some test cases. I can have this ready somewhere this week or early next one.
Do you want to include it in 2.40?

mpvader, 2019-11-04:
No; too late for that I’d say. V2.50

jepefe, 2020-01-14:
Done, tagged and added to the v2.60 ToDo.

The set of conditions to optimise PV power is not included, it is far more complex and it is out of the scope of the required change to get battery readings from systemcalc.

This issue can be closed as soon as it is included in Venus.

closed 2020-01-16 by jhofstee