victronenergy/venus #520: Cleaning up to Settings -> System setup menu

mpvader, 2019-10-07:
Its has grown a bit out of control, see below.

- [ ] AC input 1 <- show this only when main-vebus-service/NrOfAcInputs > 0 (so hide it for an inverter)
- [ ] AC input 2 <- same, but then when NrOfAcInputs > 1 (so hide it unless its a Quattro system)
- [ ] Grid loss alarm -> Rename to Monitor for [grid failure|shore disconnect]; and hide when there is no AC input configured as grid or shore.
- [ ] Move all four status items (Synchronize VE.Bus SOC with battery, Use solar charger current to imporve VE.Bus SOC, Solar charger voltage control, Solar charger current control) into a sub menu, called "Status" at the bottom, and visible only when access level is Super user. Why? Since by now we know the code works well and this is all about very old firmware versions etc.
- [ ] Move all DVCC Items in its own menu: Settings -> DVCC. 
- [ ] Shorten the explanatory text to "Read the manual before enabling this feature". (And we'll be making #519 also in v2.40)
- [ ] Only show the `Temperature Sensor`, and `Used temperature sensor` in case `Shared Temperature Sense` is enabled.
- [ ] Delete the BMS Control item, its no longer necessary.

[image]

mpvader, 2019-10-07:
## Preview of the new System setup menu:

[image]

## Preview of the new DVCC menu:

[image]

jhofstee, 2019-10-08:
gui v4.2.1 includes this

closed 2019-10-08 by jhofstee