victronenergy/venus #525: Cannot set ip address / gw / dns on Color Control GX GUI

avs009, 2019-10-29:
Hi, 
The GUI interface config of Ethernet in Color Control GX not allow to add the following IP address parameters:
IP: 189.106.216.69
mask: 255.255.255.240
Gateway: 189.106.216.65
DNS: 189.106.218.195

The problem occurs when try change the second portion in 3 digit upper than 5, they change for 0.

My workaround is set they direct with ssh, because i need to set static ip.

**IPv4 Valid ranges (According windows network manager):**
First portion: 1 to 223;
Second, Third and Fourth portion: 0 to 255; (including 006, 007, 008, 009, 016, 017, 018, 019,...)

jhofstee, 2019-10-29:
When making an issue, do report the version involved, that is a requirement. Closed by lack of that but it can be reopened of course.

avs009, 2019-10-30:
Sorry, i forgot this vital information.
Tested in versions:
2.32
2.33

avs009, 2019-10-30:
> When making an issue, do report the version involved, that is a requirement. Closed by lack of that but it can be reopened of course.

@jhofstee  Can you reopen this issue?

mpvader, 2019-11-03:
Hi @avs009 ; thanks for the versions. Could try our latest beta? We fixed (what I think is) exactly the bug you're describing here.

https://community.victronenergy.com/questions/32924/venus-os-v24037-available-for-testing.html

avs009, 2019-11-04:
> Hi @avs009 ; thanks for the versions. Could try our latest beta? We fixed (what I think is) exactly the bug you're describing here.
> 
> https://community.victronenergy.com/questions/32924/venus-os-v24037-available-for-testing.html

Yes, i will try test this beta tomorrow.

avs009, 2019-11-08:
> Hi @avs009 ; thanks for the versions. Could try our latest beta? We fixed (what I think is) exactly the bug you're describing here.
> 
> https://community.victronenergy.com/questions/32924/venus-os-v24037-available-for-testing.html

Hi Again @mpvader,

Yesterday I tested the beta 2.40~37, you're right there is no more this bug.

mpvader, 2019-11-08:
Good! Thanks.

Closed

closed 2019-11-08 by mpvader