victronenergy/venus #532: Missing /opt/victronenergy/vedirect-interface

bigalnz, 2019-11-12:
While following the instructions at : https://github.com/victronenergy/venus/wiki/raspberrypi-install-venus-packages and the step where configure vedirect

mkdir /data/services
cp -r /opt/victronenergy/vedirect-interface/service /data/services/vedirect00
rm /data/services/vedirect00/down
rm /data/services/vedirect00/log/down

The directiory to cp vedirect-interface is missing.

jhofstee, 2019-11-12:
and did this step `apt-get install vrmlogger` succeed?

bigalnz, 2019-11-12:
Yes.

bigalnz, 2019-11-13:
`pi@raspberrypi:~ $ sudo apt-get install vrmlogger
Reading package lists... Done
Building dependency tree
Reading state information... Done
vrmlogger is already the newest version (2.131-r0.0).
0 upgraded, 0 newly installed, 0 to remove and 15 not upgraded.
pi@raspberrypi:~ $
`

bigalnz, 2019-11-14:
Anything else that I can check?

bigalnz, 2019-11-19:
> and did this step `apt-get install vrmlogger` succeed?

Any other ideas?

bigalnz, 2019-11-19:
vedirect-interface doesnt even seem to exist as a repo????

danielparnitzke, 2020-04-11:
Any news here? I have the same problem.

dpackham, 2020-07-22:
I have this issue and it is because I did not have vedirect-interface installed.  it would not install due to a libevent version that was too new.  I opened a bug on that separately here

jhofstee, 2020-08-02:
lets move these issues to https://github.com/jhofstee/meta-bin-deb, since they aren't supported by Victron itself.

closed 2020-08-02 by jhofstee