victronenergy/venus #533: systemcalc: possible offset error in `/Timers/*`

thlassche, 2019-11-14:
While working on a speedup of the production report I noticed something strange in the data of `Tgs` (Time generator, system/Timers/TimeOnGenerators). 

Sometimes a 15 minute delta contains a value of 901 or 902, while the theoretical maximum should be 900 I guess (15 minutes * 60 = 900 seconds). The same is true `/Timers/TimeOnGrid`, and most likely the other `/Timers/` as well.

izak, 2019-11-14:
This is normal, or at least expected based on the current implementation.

The timers are updated every 10 seconds, so the deltas may sometimes use a value that is on average 5 seconds old. So this value might end up being anything from 900 to 910.

Fixing it on this side would require updating the counter once a second. And even then it could be off by one second based on how things ends up being timed.

thlassche, 2019-11-14:
@izak Ok, thanks.

closed 2019-11-14 by thlassche