victronenergy/venus #534: VNC password lockout

izak, 2019-11-14:
On devices without physical buttons (Venus-GX, Nanopi), a forgotten VNC password will lock you out. If remote support is also off, the user will have to do a full reflash of the device to regain access to it, since we cannot remotely unlock it either.

Ideally it should be possible to unlock the device with some kind of button combination, or if not, we need to document ways that a user might try.

One way might be to prepare a special file to be placed on an sd-card that will unlock the device during boot, using the data population and rc.local functionality already in Venus.

mpvader, 2019-11-14:
For the Venus GX, v2.40 has a fix: the button will get this function. Iirc this is already implemented in v2.40

izak, 2019-11-16:
This turns out to be a duplicate of #451.

mpvader, 2019-11-16:
Except that the nanopi based boards dont have a pushbutto ; I’ll leave it open

jhofstee, 2019-11-16:
they don't? There is one on the character display print. Not sure how that should function though.

mpvader, 2019-11-16:
Well; actually I didnt type what I was thinking: I am not sure about the MP-II GX etc boards; so leave it open to be looked i to for those. To complete another day

somombo, 2019-11-16:
Wow this issue was reported two days ago, Thanks for posting this up @izak... 

Me and a victron distributor (in zambia), have been racking our heads about how to get myself unlocked for the past couple of days. I have a multiplus ii gx and I must have mistyped the password during setting.. Couldn't, for the life of me, figure out what the typo might have been.. 

The victron distributor finally managed to reset my password using an rc.local file (I shall post up more details later)

somombo, 2019-11-16:
On MultiPlus II GX the following prompt is *very* confusing/misleading to the enduser. It says if you forgot the password you should enter it on the device. But there is no physical way to be able to enter the password on the actual device (MultiPlus II GX). This prompt needs to be updated with the correct guidance.

[image]

izak, 2019-11-16:
@somombo I'm glad to hear it is resolved for you. You are actually the reason this issue was created in the first place, though of course you don't know how the request traveled behind the scenes and eventually ended up with a tech (me) who also provided the rc.local workaround.

Thanks for responding, and we'll leave it open until we can come up with a solution for the nanopi-based units.

somombo, 2019-11-16:
Also in my case, I didn't technically forget the password that I set.. I think I merely mistyped it and casually hit save/okay expecting that the system would prompt a password confirmation. It would alleviate alot of stress for many if there was just an "Enter password again for Confirmation" prompt, so that cases of mere mistyping can be almost nearly eliminated. Please consider adding this as a feature in a future firmware update

somombo, 2019-11-16:
Wow @izak what a small world it is! indeed.. Thank you so much for your timely effort. And being on top of things!

somombo, 2019-11-16:
## Reset Button MP-II GX
> Except that the nanopi based boards dont have a pushbutto ; I’ll leave it open

> Well; actually I didnt type what I was thinking: I am not sure about the MP-II GX etc boards; so leave it open to be looked i to for those. To complete another day

@mpvader I believe MP-II GX contains a [nanopi neo core-lts](https://github.com/victronenergy/venus/wiki/machines) based GX device. As shown in the pic below there is in fact a physical "reset button" on MP-II GX... It is reminiscent of what you'd find on your router.. Unlike a common router reset button however. Holding this button down for 30 to 60seconds seems to merely reboot the GX device 

It would be ideal if holding down the reset button on that device would pretty much factory reset it (like how it does on a wifi router).

While we are on the topic of the reset button.. I found it frustrating that the help in the MP-II GX's online manual in the table in [(Appendix A - Connection Overview, on the reset button row)](https://docs.victronenergy.com/manual-multiplus-ii-gx-24Vand48V-3000and5000-230V.html#appendix-a-connection-overview) was pretty confusing and not useful.. All it says is 

>Reset Button | What is the reset button used for? MCO

Please consider correcting this.

[image]

mpvader, 2019-11-17:
Thanks for your comments @somombo!

We’ll clear it all up.

jhofstee, 2019-11-17:
it just power cycles the nanopi print. As in it reset the whole print, but not the settings etc.

somombo, 2019-11-17:
> it just power cycles the nanopi print. As in, it resets the whole print, but not the settings etc.

@jhofstee by 'nanopi print' I assume you are referring to the blue display alone? I personally can't imagine when power cycling the display would be terribly useful. However, some way of **physically** "factory" resetting the GX would be very useful, in my opinion.

jhofstee, 2019-11-17:
It is clear what you are looking for. What I tried to explain is that the reset button won't help, it just resets all peripherals, like the screen. But settings will remain the same. This is a hardware thing so it cannot be made to do something else...

somombo, 2020-06-12:
any update on the status of this issue?