victronenergy/venus #537: GX menu structure is illogical

petajoulecorp, 2019-11-21:
2.40~37 on CCGX here

The GX menu structure has probably grown historically/organically and it shows.

Settings for the GX device itself are intermangled with settings for the Installation as whole, which - by the way - is wrongly called "System" and not "Installation".

There should be a 

    Settings > GX Device

For all the parameters, configuration etc. of the GX Device itself. And

    Settings > Installation

(not "System") for the overall installation. 

In the `GX Device` menu structure, there would be stuff like

```
General
Date & Time
Display & Language
Ethernet
Mass Storage (!)
Relay
```

etc.

The `Installation` should contain all regarding the overall Installation: PV Inverters, Punps Tanks, VE.Direct, VE.Bus devices etc.

----
Right now, the illogical mixup, where e.g. Mass Storage handling is hidden under VRM online portal causes problems handling these devices independent of VRM settings.

There should be a clear distinction of parameters that concern the GX device itself and parameters that concern the installation (i.e. something the GX device only monitors/watches over. The environment it is placed in - if you so will). 

According to the "Theory of mind", intelligent children develop from an age of 15 months a concept of "me" and differentiate it from "the environment". Some take their time up until 24 months. I believe, after more than 6 years GX could slowly develop that skill too. ;-)

jhofstee, 2019-11-21:
Closing this, "where e.g. Mass Storage handling is hidden under VRM online portal causes problems", that is an existing issue. We are not going to add a "GX Device" under setttings since almost all settings will end there, just causes more navigation.

closed 2019-11-21 by jhofstee

petajoulecorp, 2019-11-22:
Very well. We still believe the raised issue is valid:

current GX menu structure does not logically well separate the device from the (surrounding) installation.

One can agree that

```
Settings > GX Device
Settings > Installation
```
Would add unnecessarily one depth of navigation to the tree:

https://www.victronenergy.com/live/_media/ccgx:menustructure_ccgx.jpeg

Which is why this distinction could be made on the topmost level already:

[image]