victronenergy/venus #538: handling microSD / external storage has several flaws

petajoulecorp, 2019-11-21:
2.40~37 on CCGX here.

see also #537

The whole handling of external storage under

    Settings > VRM online portal

and not just `Settings` - or `Settings > GX Device` brings several problems.

First off, it's puzzling from an usability point of view, as this is a hardware configuration topic of the GX device itself, it's logically equivalent to e.g. setting up Ethernet, Display and Language, Firmware settings etc.

Second, documentation under https://www.victronenergy.com/live/ccgx:start#datalogging_to_vrm
seems obsolete and is misleading at best.

Under 
[image]

it refers to a menu point "logging to a microSD card or USB flash drive", which is not existant. It further refers to a "chapter"(sic! should be: "section") 2.2 for more information, which is not existant either: next section (after 2.1) is 2.4

What's worse, if you disable logging, there is no way to actually issue the "eject safely" command as all menu points for external storage are inactive or not present. Of course if you simply eject the device under these circumstances, you get the warning about unsafe eject.

All in all, placing the mass storage handling under the VRM portal complex - together with the documentation flaws mentioned above, makes use of external storage ... cumbersome.

jhofstee, 2019-11-21:
mostly duplicate of https://github.com/victronenergy/venus/issues/75

jhofstee, 2019-11-21:
I removed the mentioned text. closing as duplicate.

closed 2019-11-21 by jhofstee

mpvader, 2019-11-25:
Hi Petajoule, we fixed the mistake in the manual. Thanks for notifying us. As you've seen, we agree that the sdcard handling is not perfect. But we dont think there is a quick fix, and prioritise other changes higher up; hence no changes any time soon.

At some point we will redo the whole menu; awaiting that.

kevinvh001, 2021-06-26:
Sometime in the past month:

1. My CCGX has stopped recognising my MicroSD card as external storage under VictronConnect/Settings/VRM Online Portal/MicroSD/Usb - it shows "No storage found".

2. VictronConnect will no longer find my Color Control GX device under Devices/Local.

Everything worked fine before I recently upgraded to VictronConnect V5.42 and CCGX V2.71. I don't know when the above stopped working.

VictronConnect/Settings/VRM Online Portal/MicroSD/Usb:

I tried two seperate MicroSD cards (both previously worked) - both not recognised by CCGX. Tried reformatting both MicroSD cards (16GB and 32GB Fat32) - not recognised by CCGX. Inserted a 16GB Fat 32 USB stick in the CCGX USB port - recognised by VictronConnect/Settings/VRM Online Portal/MicroSD/Usb as external storage.

VictronConnect not showing CCGX in Devices/Local:

If I switch to the VRM tab under Devices, then VictronConnect will recognise the CCGX and show it as 'online' and allow me to login to the VRM Online Portal on my Web browser. However, it does not show the "Remote Console" tab and I cannot connect the CCGX with my browser (as I could before).

Has anyone else had these issues since upgrading VictronConnect or the CCGX firmware?

Anyone have suggestions about how I can resolve my issues?

mpvader, 2021-06-27:
Hi @kevinvh001, for the sdcard, you can best roll back to v2.66 by the sound of it, and wait for more info from us. That will be here: https://community.victronenergy.com/questions/93185/venus-os-v2722-available-for-testing.html in the thread by Kevin Windrem.

VictronConnect: we’ll look into that, and reply to you on community, on that same page.

thank you reporting.

ps: this is a really old issue; lets not keep talking on this one - thank you.