victronenergy/venus #543: make dbus-mqtt connect to different mqtt server based

mpvader, 2019-11-28:
For load balancing, we’ll host multiple mqtt servers.

There is a defined algorithm on how to know which one to connect to. That algorithm takes the vrm portal id as an input.

The affected mqtt link is dbus-mqtt. Not mqtt-rpc.

mpvader, 2019-11-28:
Either Teun or Wiebe can make the (I suppose small) change in the python code. Who of you two makes it doesnt matter to me :-)

thlassche, 2019-11-28:
First proposal of an implementation over here: https://github.com/victronenergy/velib_python/pull/20/files
@wiebeytec @mpvader

wiebeytec, 2019-11-28:
> The affected mqtt link is dbus-mqtt. Not mqtt-rpc.

Just for completeness here: the submodule velib_python _does_ need to be updated in mqtt-rpc, because otherwise they will overwrite each-other's result.

thlassche, 2019-11-30:
Merged and tagged it in mqtt-rpc and dbus-mqtt.

Added it to the v2.40 todo: https://github.com/victronenergy/venus-private/wiki/todo-v2.40#v240---done---waiting-for-recipe--venus-maintainer-add-new-ones-at-the-top-of-this-list

Happy to know if I did something wrong in the tagging and listing it on the todo

mpvader, 2019-11-30:
I’ve informed community: https://community.victronenergy.com/questions/35016/edit.html

mpvader, 2019-12-03:
Added to Venus in v2.40~44

wiebeytec, 2020-02-04:
It's in the works to make VictronConnect use the new servers (I added it to the checklist). I already tested it using `/etc/hosts` (together with a test site), and it works (as do VRM live updates).

What's the status of the Venus Docker Grafana? Once VictronConnect is done, that would be the only block, as far as I can tell.

mpvader, 2020-02-05:
Venus docker grafana is already dine and released. I ticked the box

mpvader, 2020-04-16:
All done - closing this.

Welcome to reopen @wiebeytec or @thlassche if necessary.

closed 2020-04-16 by mpvader