victronenergy/venus #544: Cerbo GX: make sure VE.Bus panel detect is floating

mpvader, 2019-11-28:
instead of hard-shorted to ground like in the CCGX.

Relates to #106 and a whole lot more.

vejpasop, 2019-11-29:
In order to prevent confusion I would suggest to change the title to "Make sure VE,Bus panel detect line can be controlled".
This was tested successfully in the A6 revision of the Cerbo. I just verified the schematics of the A9 also. Nothing changed there. So this can be closed.

mpvader, 2019-11-29:
Ok; and what does this mean for customers?

And how does it compare to CCGX, Venus GX, Octo GX and others?

Do we need to clean up documentation somewhere wrt this?

vejpasop, 2019-11-29:
It fixes https://www.victronenergy.com/live/ccgx:ccgx_faq 
**"Q20: Issues with Multi not starting when CCGX is connected / Caution when powering the CCGX from the AC-out terminal of a VE.Bus Inverter, Multi or Quattro"**

I don't recommend powering the Venus device through the AC out so I would not advertise this as a cool new feature.

mpvader, 2019-11-29:
> And how does it compare to CCGX, Venus GX, Octo GX and others?

vejpasop, 2019-11-29:
CCGX, Venus GX: panel detect fixed to ground
Octo GX, Cerbo GX, Canvu500 and other MK3-USB based devices: panel detect controllable

closed 2019-12-19 by mpvader