victronenergy/venus #548: dbus-fronius: Fronius PV Inverters sunspec scan times out

izak, 2019-11-29:
## Background
When scanning for Fronius PV-inverters the GX device will first attempt to use Fronius SolarApi. If a Fronius inverter is found, it will use the information obtained from SolarAPI to scan for sunspec support. The timeout for a sunspec probe is 5 seconds. If sunspec support is found, the "retrieval mode" changes from SolarAPI to SunSpec.

A periodic rescan is done to find any new inverters, or old ones that may have been off or otherwise unreachable. If upon a rescan SunSpec support is not detected, the retrieval mode changes back to SolarAPI.

## Problem description
Datamanagers with more than a few (observed with 4) PV-inverters become slow enough that the Sunspec probe will time out. This causes PV-inverters to randomly switch back and forth between SolarAPI and SunSpec.

It also causes the service to disconnect from D-Bus and reconnect, causing a large amount of configChange logging.

izak, 2019-11-29:
The probe timeout was increased to 15 seconds in dbus-fronius 1.40.

jhofstee, 2019-12-16:
reopen since it is reverted for now, see https://github.com/victronenergy/venus/issues/568

closed 2020-01-16 by jhofstee