victronenergy/venus #549: dbus-fronius: Intermittent zero power values are logged that are not real

izak, 2019-11-29:
[image]

Reported by Simon Hackett.

Investigation shows that the data retrieved via modbus-TCP (SunSpec) contains spurious zero-frames. This happens for SunSpec models 101 and 103. The entire frame is zero except for the Operation State, which contains the value 7 (Fault). The fault clears almost immediately. The same zero-value is not shown on Fronius's solarweb platform.

Currently waiting for a response from Fronius.

mpvader, 2019-12-25:
The response:

A Modbus response that contains only zeros except for the two registers “operating state” and “operating state vendor” is possible in case there is a Solar Net communication issue (Datamanager to inverter communication).
This can happen either because of a software bug but also for example in case the Solar Net communication is simply too busy (in case too many inverters per Datamanager, inverter updates running, too many Modbus queries, etc…) or simply in case of bad cabling.

In Izak’s case (one inverter with DM card inside, no cabling issues possible in that case) it was a software issue that was fixed with an early fro30xxx.upd firmware version (Izak's Primo had fro29390.upd installed).

In case of power commands, the Recerbo sometimes lost or did not reply to Solar Net queries because of waiting for a response of the power stack first to make sure the last power setpoint is stored correctly. This problem is fixed with the latest official firmware version (currently fro30381.upd).

As there are inverters out in the field that are never updated because the Datamanager is not in Solar.web but they are often online in VRM, and knowing this issue can happen again because of bad cabling for example, we recommend to discard these “bad” messages in software, meaning a response that holds “operating state” 7 (= fault) and “operating state vendor” 10 (and also the two additional Solar Net related values 9 & 11 could be considered here), like in the case Izak reported, and re-trigger a new query or use the next scheduled correct message instead.

The “operating state vendor” can hold the following values:
1: Off
2: Sleeping
3: Starting up
4: Tracking power point
5: Forced power reduction
6: Shutting down
7: Fault
8: Standby
9: No SolarNet communication    –    in case of bad Solar Net cabling this can happen more or less often
10: No communication with inverter    –    in case the inverter does not respond with correct values for any reason (like in this investigation)
11: Overcurrent on SolarNet plug    –    Solar Net is permanently down, no data transmission possible
12: Inverter is being updated
13: AFCI

The values 9, 10, 11 will always come with an additional "operating state" of 7.

mpvader, 2019-12-25:
For large systems (multiple three phase fronius inverters, all on one datamanager) we noticed that is quite likely for the datamanager to be overloaded and thus respond with thus 0 messages.

Solution is underway by Izak and currently being field tested.

closed 2020-01-16 by jhofstee