victronenergy/venus #550: dbus-vrm: Log interval setting seems to have an off-by-one error

thlassche, 2019-11-30:
See: https://community.victronenergy.com/questions/34901/vrm-log-interval-12h-and-24h.html

Checked it on VRM:

[image]

thlassche, 2019-12-13:
It's a bug in the GUI, only for 12 hours and 1 day, see: https://github.com/victronenergy/gui/blob/37d3cd6044fc8be5920c4afd8d6cffe63798d7fd/qml/PageSettingsLogger.qml#L39

- 12 hour is set at 7200s which should have been 43200s
- 1 day is set at 14400s which should have been 86400s

FYI @wiebeytec

mpvader, 2019-12-13:
Teun wrote:

> Only affects the 12h and 24h setting.

I checked 1 minute and 15 minute interval on v2.40; and they work fine. 1 minute = 60 seconds, and 15 minute = 900 seconds (duh)

jhofstee, 2019-12-19:
reopen since this can go in v2.41 as well

mpvader, 2019-12-19:
In v2.41, closing

closed 2019-12-19 by mpvader