victronenergy/venus #551: make Garnet Seelevel senders (N2K version) work

mpvader, 2019-12-02:
> The problem as mentioned early in this thread is that the SeeLevel NEMA2000 sender reports 3 tank sensors and the code within Venus assumes only one sensor per NEMA2000 device. So the display switches between tanks as the SeeLevel system reports values for each tank.

https://community.victronenergy.com/questions/6170/rv-c-can-bus-tank-monitors.html?childToView=35196#comment-35196

This issue is currently waiting for a sample.

mpvader, 2019-12-16:
I reminded Garnet wrt shipping us the senders today

mpvader, 2020-01-28:
we have received the senders, they are in the Groningen office. Next step is for Jeroen to look into supporting it (their N2K type sender, not the RV-C one).

kwindrem, 2020-03-20:
Any progress?

mpvader, 2020-03-21:
Not yet no, sorry

mpvader, 2020-04-15:
There is progress, we have a working test setup and such. But we need to make a change to our logic thats a bit more involving than most changes. Our code now expects each device instance to send one type of data. And what the Garnet does is different: one device instance sending three tank levels, each with their own data instance.

kwindrem, 2020-04-27:
Here is some information about the SeeLevel messaging that may be of use in your native support of SeeLevel tank systems.

SeeLevel reports status on their display consistent with /Status reported by other tank sensors:
Disconnected, Short, etc. I hope this status can be displayed on the Mobile Overview Tanks tiles and on the Tank device pages.

According to Tom Stalker (Garnet marketing), SeeLevel reports tank status as an out of range Level. The current N2K driver in Venus is limiting /Level to 0% - 100%  and is not updating /Status.

I was not given the actual Level values sent for the various status conditions.

During debug of my tank repeater mechanism, I discovered that SeeLevel gets into a state where N2K tank level messages are sent very close together. All three messages are sent in a 3-4 second period and when SeeLevel first starts, the messages are roughly equally spaced. After a while, sometimes as much as a couple of days, the messages bunch up.

preeve9534, 2021-02-07:
Just dropped in on this thread and I have some thoughts.

About four years ago I reported to Victron support this exact issue with a Maretron TMP100. I have been revisiting Victron's various channels on and off since then hoping for a solution.

1. The problem discussed in this thread may be a generic issue with the CCGX and multi-tank sensor reporting. I know that the Maretron TMP100 is NMEA 2000 compliant in its use of instance numbers as tank differentiators on reported data. I don't know much about the Garnet device but I suspect it may also be N2K compliant.

2. If my assumption about Garnet is correct, then all well and good. If not, then fixing this problem for the Garnet device (as this issue's title and the discussion above implies is happening) is just plain daft. Surely, the issue needs fixing so that the CCGX becomes N2K compliant in its handling of tank data and hence supports all N2K compliant devices?

3. If (1) and (2), then in my view this issue should be either renamed so that it accurately represents the problem (instead of misleadingly associating it with one manufacturer's device) or closed and directed to a new issue that serves the same function.
Something like "Broken support for multi-channel (multi-instance) tank sensors"? Maybe people will be better able to land on this discussion whatever their hardware allegiance?

4. Four year's ago I was enthusiastic about the CCGX's vaunted ability to display tank data and I am a little sad that this is a still a problem.  I have five tanks and at the moment and, patience elapsed, I am about to write a plugin to re-map my tank sensor data in Signal K so that it appears to come from five discrete sensor modules: crazy!

5. Is there any more news on when an N2K compliant firmware fix might be available? I am beginning to worry that my CCGX will die before it ever displays any meaningful tank data ;-).

Paul

mpvader, 2021-02-07:
Hi @preeve9534 , thank you for the feedback. The Maretron TMP100 is a 6-input Maretron temperature sensor. We don't support reading of any N2K temperature sensors at the moment: perhaps you mean a different Maretron device?

It is frustrating, also for us, that fixing this issue (supporting a single N2K device that measures multiple tank levels) takes forever; but please keep in mind that our products are first and foremost for monitoring and controlling a power system. We do what we can to monitor more; but as said above this particular change is unfortunately rather a big one in our code structure. So tends to be delayed in favour of other projects. It what it is I'm afraid and I don't expect a solution any time soon.

SignalK could be a relatively simple way to make it work indeed. Instead of resending the data out to N2K, remapped, the alternative way to do it would be to expand [the Venus OS SignalK plugin](https://github.com/sbender9/signalk-venus-plugin) to send data to D-Bus directly. And just in case you're not aware, make sure to read this: https://www.victronenergy.com/live/venus-os:large

preeve9534, 2021-02-07:
Thanks for the response.

Apologies for misquoting the Maretron product number - I meant to say
FPM100.

I completely understand Victron's position on tank level monitoring not
being "core" business:
I imagine that you now regret raising some customer's expectations all
those years ago!

I still think you should rename this issue to something sensible so that
other folk interested in
tank level support can find this discussion more easily and so be better
informed.

Thank you as well for the D-Bus and :large suggestions. For me, though
perhaps not for others,
the re-mapping in Signal K makes more sense: my ship runs under Signal K
and I try to keep
hardware devices like the CCGX in stock configuration as far as I can.

I'll publish my Signal K plugin as an npm in due course and put a link on
the end of this
thread in case it helps anyone else.

P

kwindrem, 2021-02-07:
Paul,
I have developed a "repeater" that runs on Venus OS that allows multi-sensor tank systems to display as separate tanks. It was written for the Garnet SeeLevel N2K system but should work with other devices as well. Some tweaking may be necessary.

You can find my repeater here:
https://github.com/kwindrem/SeeLevel-N2K-Victron-VenusOS

Garnet's SeeLevel product line includes an NMEA2000 version but you need to contact them for a unit since it's not in their catalog.

preeve9534, 2021-02-07:
Ah yes - I spotted that earlier today and thought "Good man!".  Thank you
for the heads up though.

I did an install and started to play, but then ran into this:

https://github.com/kwindrem/SeeLevel-N2K-Victron-VenusOS/issues/1.

Haven't had time to delve into the code yet,

Paul

preeve9534, 2021-02-10:
I now have a working (almost complete) dbus interface to Signal K which supports multiple tanks.

https://github.com/preeve9534/venus-signalk-tank-service/blob/main/README.md

This leverages the GUI modifications implemented in @kwindrem 's project mentioned above.
@kwindrem - hope you don't mind me using your work in this way?

Paul

mpvader, 2021-02-10:
Hey @preeve9534 that looks like well done! I had a brief look through the code, two thoughts, which you're probably aware of anyway:

1) if this would be a signalk-plugin, then it would be easy to add onto Venus OS Large. (and easy to install for any ser having SignalK)
2) I dont see /CustomName in your script. You could use that to forward any configured name of the device onto Venus OS.

ps. I'll look into renaming this issue. I still like Garnett in the same since thats what many people are looking for and this way they can easily find it.

preeve9534, 2021-02-10:
@mpvader thank you for the feedback.

1. Good point re Venus OS Large and Signal K plugin. Because I have a CCGX
I didn't go there...
2. Good idea. I'll add that.

Maybe add "Maretron FPM100" to the title then? I think boaty types will
have difficulty finding this thread because SeeLevel N2K is not widely
known. in that community.

P

mpvader, 2021-02-10:
@preeve9534 , just to confirm: I've updated the title as well as the first message in this issue.

Regarding the customname, what we usually do is read it from (and store it in) PGN 0x1F016, see also here: https://github.com/victronenergy/venus/wiki/dbus-api.

ps. are you on the SignalK slack?

preeve9534, 2021-02-10:
@mpvader looks good ... except should be FPM100 not FMP100 ;-)

HUA on customname.

Active on Signal K github, but not slack... thinks...

P

mpvader, 2021-02-10:
http://slack-invite.signalk.org/ - perhaps see you there; I've a question, but would be good to stop spamming the people that get email updates everytime we type somthing here.

mpvader, 2021-05-20:
Hi all, per v2.70~12; available as a release candidate, this issue is finally fixed and closed!

closed 2021-05-20 by mpvader

mpvader, 2021-06-22:
Careful, a few expected features are not included. Definitively custom names are not. And maybe there is something with tank types as well, but could also be that works fine: I don't know and didn't test.