victronenergy/venus #557: Improve viewing of tanks on the overview page

mpvader, 2019-12-10:
From the customer:

> the Victron display has some very user-friendly power pages. Unfortunately it suffers from having the same user-friendly tank pages and I think this is needed.
> This could be done with a “Tank” button on the main page.
> Pressing this will change the display to show all water tanks on one page and with the possibility to switch further to other tanks or back to the main page.

[image]

We'll have to see how to consider the CCGX (non touch); and there will be other constraints as well. So I don't know if we can or should do exactly like drawn. But, I promised the customer to come up with a good solution. And with deadline for final release of Q2 2020.

jepefe, 2021-02-25:
### New tanks overview page:

[image]

- Tanks overview can be enabled on the display settings page, same as mobile overview.
- Six tanks per fluid type
- Filtering per fluid type
- All tanks are shown together when the total number of connected tanks is less or equal to 6, filtering per fluid type is still possible
- Touch & button navigation

Watch the video to see the behavior:

https://user-images.githubusercontent.com/6424631/109156536-3fc7ca80-7771-11eb-9eab-a97a12ce409c.mp4

### Improved mobile overview tank tiles:

Due to the limited available space, current mobile overview can only properly show up to four tanks:
[image]
[image]

With the changes, when there is not enough space, the tiles are compacted to make possible to fit up to ten tanks. 
With the compact tile layout, the tank level bar and the tank name share the same row so the space for the name is reduced, this has been solved by applying a text scroll every 15 seconds. 

Tank custom names are now displayed when available, otherwise the fluid type is shown. Requested in: https://github.com/victronenergy/venus/issues/715

[image]

Watch the video to see the scroll in action:

https://user-images.githubusercontent.com/6424631/109159007-663b3500-7774-11eb-91cc-a968342e4e34.mp4

All the changes are implemented and available in gui:wip_tanks_overview and themes:wip_tanks_overview branches. 

@mpvader please take a look to the screenshots and videos and let me know what do you think.

kwindrem, 2021-02-25:
Very nice. 

For lots of tanks, I like your compressed display. It would be great if that tile could kick in only when necessary however.

I don't know if you've seen my tank U/I  mods:
The first is part of the package that splits NMEA2000 tank systems into separate displays. 

Of interest here is it includes a compressed tank tile that allows 6 tanks to be displayed on the standard Mobile Overview. The tank tile also displays fluid level in absolute units (e.g., gallons) in addition to percentage.

https://github.com/kwindrem/TankRepeater

The other change of interest is an Enhanced Mobile Overview that hides the Pump switch if it's not defined.  (There are actually 3 different GUI mods in this package which you can choose to install independently.)

https://github.com/kwindrem/GuiMods

jepefe, 2021-02-26:
Hi @kwindrem, I think I saw a modded mobile overview but I can't remember where, perhaps in community?
I'd like to see your mods,  do you have a link with screenshots? Please don't post the screenshots directly on this issue.

kwindrem, 2021-02-26:
Sorry, I thought there was a screen shot in the GuiMods repo but it wasn't there. Is now. Here's a copy
[image]

preeve9534, 2021-03-03:
Some thoughts on this very welcome discussion.

1. The introduction of icons, bordered buttons and rounded corners in the proposal
    https://github.com/victronenergy/venus/issues/557#issuecomment-785890425
    represents clear shift in GUI presentation style from the Venus default.

    Is there a Victron design / style guide or any desire to conform to the de-facto
    Victron default style?

2. The N2K standard defines seven fluid types. Relying on icons only as selectors
    of fluid type sub-pages raises the issue of how are (gasoline and diesel) and
   (grey water and black water) to be unambiguously differentiated by symbol alone?

3. My recollection is that the N2K fluid types are 0 = Diesel, 6= Gasoline - use of
    the term "Fuel" is ambiguous.

4. The metric abbreviation of litre is 'l' not 'L'.

5. Adding a configuration filter for eliminating tanks with a particular Venus DeviceInstance
    would allow the user who has multi-channel tank sensors (like the Maretron FPM100)
    which have broken support in Venus to be filtered out of the tank display which they
    will otherwise break.

P

jepefe, 2021-03-08:
@kwindrem
Thanks for sharing your ideas, I borrowed the idea of hiding the pump button when not enabled. And not now but in the future we'll also show the remaining value in the selected units as well.

@preeve9534
Thank for your remarks:

1. No, there isn't a design/style guide available.
2. We currently only support 6 fluid types and Fuel (Gasoline) is not one of them, only generic Fuel(0) is available. We will fix this. As you can see in the screenshots and videos, besides the icons, there is a text description for each fluid type at the top of the page so that will help in case someone doesn't understand the icon meaning. Anyway I think that with the current icons it is easy to differentiate the fresh, waste and black water fluid types, see the attached screenshot.
3. The SI states that the unit should be written in lower case if it doesn't derive from a person name, but as an exception, "L" is allowed for liters as it can be easily confused with the number one "1". This is why we use "L" instead of "l"
4. Filtering by device instance is not just a UI improvement and requires deeper changes, so it is out of the scope of this issue.

[image]

jepefe, 2021-03-08:
Since my previous commit with the screenshots:
- Tanks overview can now properly show up to 10 tanks per fluid type.
- Pump relay button is hidden when function is not enabled.

Added to the v2.70 ToDo.

preeve9534, 2021-03-08:
https://github.com/victronenergy/venus/issues/557#issuecomment-792699749

These new pages are looking really nice and I would love to be able to use them but, like others with ship-based multi-tank systems, will not be able to.

I understand your response on point (4) in my earlier post and I absolutely accept that correcting the multi-channel tank sensor data acquisition problem is outside the scope of this project. However I do not agree that dealing with the presentation of rubbish data is outside the scope of a GUI project like this.

I think it is not unusual or unreasonable but actually quite normal and desirable for a GUI to detect faulty or exceptional data, report it to the user and, maybe, allow the user to take action about how the faulty data presents in the GUI.

In the case of the multi-channel tank sensor fault the presence of faulty data is signalled by multiple tank updates appearing within milliseconds on the same Venus tank device. This update frequency is well outside the N2K specification for maximum tank sensor update interval and in that sense at least might warrant recognition by the GUI, indication to the user and perhaps some mechanism for automatic / user-mediated suppression of data from the out-of-spec device in the GUI.

mpvader, 2021-03-08:
Hi Paul, I’m sorry, but for Venus OS this is out of spec. I know many people want their multi-tank tanksender to work properly, I do as well, and I’m sure we’ll get there some day; but not now as part part of this.

preeve9534, 2021-03-08:
OK. I'm shutting up. But.. please note that my comments aren't about getting the multi-tank sensors to work (I do after all have a solution to that), but about making the new GUI look good and feel "professional".

As things stand there will be a new "enhanced" Venus tank display that will continue to present clearly rubbish data from the broken tank sensor devices. Not a good look for Victron, I'd have thought...

closed 2021-06-04 by jhofstee