victronenergy/venus #558: Add tank senders to CI

mpvader, 2019-12-10:
We officially support a Maretron sender, and a Navico one.

https://www.victronenergy.com/live/ccgx:start#connecting_nmea-2000_tank_senders

Jeroen and Martin will know where those senders are lingering around in Groningen. If not, buy a new one.

evandromr, 2019-12-17:
The `Navico Fluid Level Sensor Fuel-0 PKr (12V)`  is now connected to the CerboGX. It is recognized with any 250 kbps CAN profile. Does not seem to support changing the fluid type or tank capacity (could it be because it is not connected to a tank sender?). 

What kind of tests do you want to run with them?

mpvader, 2019-12-18:
> Does not seem to support changing the fluid type or tank capacity (could it be because it is not connected to a tank sender?)

no, that cannot be because there is no tank sender connected. A tank sender is a mere (variable) resistor. @jhofstee did that used to work for that sender? Or not at all? If don't care too much if it works or not, as long as its clear and documented that it works (or doesnt work).

> What kind of tests do you want to run with them?

The purpose of adding this to CI is to make sure that we don't inadvertedly and unknowingly break supporting that sender: so Evandro pls connect a 50 Ohm resistor to it, and then make a script that checks that you find the equivalent in percentage or liters on the D-Bus as a com.victronenergy.tank: https://github.com/victronenergy/venus/wiki/dbus#tank-levels.

Just to refresh my memory a bit:

[image]

And here is whats on D-Bus for that sender now:
```
value@einstein:~# dbus -y com.victronenergy.tank.socketcan_can0_vi0_uc3059 / GetV 
value = {'Capacity': [],
 'Connected': 1,
 'DeviceInstance': 0,
 'FirmwareVersion': '1.3.0-MZ1657C-B1.0',
 'FluidType': 0,
 'HardwareVersion': '1.0.0',
 'Level': 0.0,
 'Mgmt/Connection': 'VE.Can',
 'Mgmt/ProcessName': 'vecan-dbus',
 'Mgmt/ProcessVersion': '2.41',
 'N2kUniqueNumber': 3059,
 'ProductId': 41312,
 'ProductName': 'Tank sensor',
 'Remaining': [],
 'Serial': []}
```

mpvader, 2019-12-18:
The setup guide; also available in google.

[Sensor_Setup-Guide engelsk.pdf](https://github.com/victronenergy/venus/files/3980066/Sensor_Setup-Guide.engelsk.pdf)

evandromr, 2019-12-30:
Added to the CerboGX, with a 56 Ohm resistor and a test script to check the corresponding level (~85%)

test stage on gitlab:
[image]

mpvader, 2019-12-30:
Hey @evandromr 👍. One question/remark: which sender did you arrange this for? The navico one? Pls then rename the ci job to be either more specific or more generic: there are more tank senders that I want to test compatibility with. And our own “VE.Can Tanl Sender adapter” is one of them. see other github issues.

evandromr, 2019-12-30:
Ok, I'll either rename or break it into parts then.
I've arranged both the Navico and our own adapter. 

Something like `cerbogx_navico_and_victron_vecan_tank_senders` looks too big to me. :sweat_smile:

evandromr, 2019-12-31:
renamed test job to `cerbogx-navico-and-victron-n2k-tank-senders`

evandromr, 2020-01-17:
I believe this can be closed

closed 2020-01-17 by mpvader