victronenergy/venus #562: Venus GX tank capacity can't be changed

mpvader, 2019-12-14:
Concerns the built-in tanks. And might or might not also concern NMEA2000 tank level senders , I didn't check.

What happens is that when you change it in the gui, it changes right back to the default, 0.200m3.

Changing it on the gui works.

I rolled back to v2.33, changed it there (that worked) and then updated back to v2.40.  The configured capacity survived the update. However, the localsettings log does show a few weird entries during the v2.40 boot, note the double slashes.

It should not be adding those settings.

```
2019-12-14 09:37:34.251482500 INFO:root:Added new setting /Settings//Tank/1/Capacity. default:0.20000000298, type:f, min:0.0, max: 1000.0, silent: False
2019-12-14 09:37:34.315032500 INFO:root:Added new setting /Settings//Tank/2/Capacity. default:0.20000000298, type:f, min:0.0, max: 1000.0, silent: False
2019-12-14 09:37:34.365333500 INFO:root:Added new setting /Settings//Tank/3/Capacity. default:0.20000000298, type:f, min:0.0, max: 1000.0, silent: False
2019-12-14 09:37:34.399323500 INFO:root:Added new setting /Settings//Temperature/1/Scale. default:1.0, type:f, min:0.10000000149, max: 10.0, silent: False
2019-12-14 09:37:34.452230500 INFO:root:Added new setting /Settings//Temperature/2/Scale. default:1.0, type:f, min:0.10000000149, max: 10.0, silent: False
2019-12-14 09:37:35.896079500 INFO:root:Added new setting /Settings//Gui/StartWithMenuView. default:0, type:i, min:0, max: 1, silent: True
2019-12-14 09:37:38.340503500 INFO:root:Added new setting /Settings//Canbus/can1/Profile. default:1, type:i, min:0, max: 5, silent: False
2019-12-14 09:37:38.594461500 INFO:root:Added new setting /Settings//Canbus/can0/Profile. default:3, type:i, min:0, max: 5, silent: False
2019-12-14 09:37:39.777892500 INFO:root:Added new setting /Settings//Vecan/can1/MainInterface/Nad. default:225, type:i, min:0, max: 251, silent: False
```

Changing the setting in v2.33 looks like this in the localsettings log:

```
2019-12-14 09:35:56.884890500 2019-12-14 09:35:56,883 INFO Setting /Settings/Tank/1/Capacity changed. Old: 0.20000000298, New: 0.289999991655
```

mpvader, 2019-12-14:
Also, the tanks don't show up on vrm for me. The temperature sensors do.

https://vrm.victronenergy.com/installation/33294/overview

[image]

Parameters logged by vrmlogger are:

```
'com.victronenergy.tank': {
	'/ProductId': {'code': 'tP', 'whenToLog': 'configChange'},
	'/FirmwareVersion': {'code': 'tv', 'whenToLog': 'configChange'},
	'/Capacity': {'code': 'tc', 'whenToLog': 'configChange'},
	'/FluidType': {'code': 'tf', 'whenToLog': 'configChange'},
	'/Level': {'code': 'tl', 'whenToLog': 'onIntervalAlways'},
	'/Remaining': {'code': 'tr', 'whenToLog': 'onIntervalAlways'},
	'/Status': {'code': 'ts', 'whenToLog': 'onIntervalAlwaysAndOnEvent'}
},
```

mpvader, 2019-12-14:
I remove the comment about tanks not showing on VRM; thats (a) a separate issue, and (b) no regression.

jhofstee, 2019-12-14:
It is a gui thing, I think we should revert https://github.com/victronenergy/gui/commit/4197130e289418c26cee4b0f3389bd0fa043649f#diff-8e60962b5e1c54ffec8e98f6ddda655e for the time being.

jhofstee, 2019-12-14:
adding @jobfeikens..

mpvader, 2019-12-14:
Ok for me to revert that

mpvader, 2019-12-19:
In v2.41~1; closing.

closed 2019-12-19 by mpvader