victronenergy/venus #563: vecan-dbus: add /Status path for tank senders

mpvader, 2019-12-14:
The VRM Portal needs one dataAttribute to be sent always. It uses it to show the age of last receiving data from the device to the user. And uses it for the list on the Device List page.

dbus-adc has a status path:

```
'0', 'OK', '0', NULL
'1', 'Disconnected', '0', NULL
'2', 'Short circuited', '0', NULL
'3', 'Reverse Polarity', '0', NULL
'4', 'Unknown', '0', NULL
```

Vrmlogger transmits that value always. And transmits the level only when there is a level.

Adding /Status, hardcoded to OK, and then in the vrm database changing the keyIdDataAttribute for tank senders from level to status would fix the Venus GX / Cerbo GX built-in tank senders not showing on the device list on VRM in case they're having issues.

Temperature is handled the same: keyIdDataAttribute is the status, not the actual temperature.

key-dataAttributes for the other devices are:

[image]

mpvader, 2019-12-14:
(slightly updated above text)

jhofstee, 2020-03-13:
The can bus change is in v2.60~9. Keep in mind that the VRM should look at level OR status for backwards compatibility.

mpvader, 2020-06-11:
note that above is changed a bit: when the tank level comes from the canbus (N2K), dbus-vecan will look at the level and set /Status to 5 (= Error) in case the level as received is invalid. In N2K, invalid is sent as MAX-INT minus one, or something like that.

jhofstee, 2020-07-10:
The error state is included in v2.60~29, VRM seems to cope with that as well.

jhofstee, 2020-07-10:
reopening this, since VRM doesn't seem to display it in the Device List when in an error state.

thlassche, 2020-07-10:
@jhofstee Fixed

closed 2020-07-10 by thlassche