victronenergy/venus #565: Possible v2.40 issue with cgwacs?

mpvader, 2019-12-14:
I have not looked into it yet, this is the info

> Hi,
>
> After upgrade firmware from 2.33 to 2.40, problems with carlo gavazi meter. Connection not stable.
> installation 9444. 2019-12-14   11.04
> 
> Back to 2.33.
> 
> Best Regards
> 
> M

izak, 2020-01-07:
I don't yet know the cause, I'm just documenting what I've done so far.

Observed symptoms: The meter is lost every few minutes. "Every few" means something between 2 and 7 minutes.

The first thing I did to test was run the new version of dbus-cgwacs (v1.90) on Venus v2.33. This combination is perfectly stable. On the evidence it seems to not be a bug in dbus-cgwcs itself, but something else in v2.40 must be acting up (kernel driver?), or it is a combination of the two. At this point I will have to obtain the user's permission to debug it in place as it seems to be specific to his setup.

mpvader, 2020-01-07:
@izak, didn't you say last time that this is due to a change in time outs of cgwacs? So probably he already had connection issues, that also were having an effect on ess performance, but only now they are actually visible to the user due to the disconnect?

izak, 2020-01-07:
@mvader I did say that if I had to pinpoint a change, in other words GUESS, then it would the the change we made in the error handling, which was one of two made for Venus 2.40.

I was wrong.

Previously dbus-cgwacs would only terminate after 5 timeouts, in other words if there are no responses. This broke when we added support for the meteo devices: Now there was a response, albeit a error response. So cgwacs was changed to terminate after 5 errors to prevent it from hanging onto a meteo device.

The timeout behaviour itself did not change.

I then postulated that if this site had occasional modbus errors already, that would explain the change in behaviour, and just starting the older version of dbus-cgwacs with debug logging will show it immediately.

But there are no modbus errors while running Venus 2.33, and both versions of dbus-cgwacs (with and without the error handling changes) work perfectly on Venus 2.33. Something else is going on. I contacted the customer to ask permission to debug deeper.

mpvader, 2020-01-07:
👍

mpvader, 2020-01-08:
Some more feedback from Micius. From old to new:

> > > Hello, I dont now you fixed this problem or not, but with 2.42 i not see problem with carlo gavazi meter.Work ok.

> > Sorry, in this worlds problem still here :(. With 2.42 loosing conection on 2-3 seconds, now 21.09 

> 2012-01-04  - 10h 26min - loosing connection.
> 2012-01-04  - 10h -28min - loosing connection
> 10-29 loosing connection
> and etc etc

izak, 2020-01-09:
Some information from debugging overnight. At certain times, there are many CRC errors on his modbus connection. In the past we ignored these errors, as mentioned earlier, which is why the problem only shows up now.

Since CRC errors could be caused by electrical noise, and such noise could be intermittent, I think it is a good idea to be more forgiving. At the moment we terminate after 5 errors in a row, but since modbus requests are spaced aproximately 250ms apart, that means we terminate in aproximately 1.25 seconds. We could allow 5 times more errors, equivalent to 5 seconds, without breaking support for the meteo sensors.

closed 2020-01-16 by jhofstee