victronenergy/venus #569: Fix behaviour with Bluetooth dongle that do not support LE

ThiemoVanEngelen, 2019-12-17:
There are currently 2 problems: 

1. Vesmart-server keeps restarting because it quits when it does not detect a valid interface offered by Bluez via the dbus ( Bluez ignores the adapter because it does not support LE) but it is restarted by daemon tools.
2. The gui shows a Bluetooth entry in services although it will not work.

Todo:

- [ ] Fix the gui that it looks at Bluez and if Bluez offers an adapter path it should show the option. It could also look at connman and if connman does contain a bluetooth technology give a warning that non LE bluetooth adapters are not supported
- [ ] Fix the behaviour of Vesmart-server to not constantly restart. This can be done in 3 ways:
  - [ ] Just keep sleeping when no suitable adapter is found and depend on the udev scripts to start and stop the service. 
  - [ ] Always start the service and let it watch the Bluez dbus services itself. 
  - [ ] Improving the start script of vesmart-server to check if the adapter supports LE.

ThiemoVanEngelen, 2019-12-17:
To improve the startup script, we could check /sys/kernel/debug/bluetooth/hci0/hci_version and check if the value >= 6 as this correspond with at least Bluetooth 4.0 as can be seen [here](https://www.bluetooth.com/specifications/assigned-numbers/host-controller-interface/).

mpvader, 2023-03-26:
Too much a nice to have to actually get it done. - apparently :-)

closed 2023-03-26 by mpvader