victronenergy/venus #578: CCGX: Remote Console enabling issues

mpvader, 2020-01-08:
An installer contacted us with complaint that I also hear from others now and then: even after having disabled the Remote console password several times (which creates the empty /data/conf/vncpassword.txt file), it still doesn't work.

I was hoping that the fix made recently in v2.42 (adding sync statement after making that file) would solve it, but this customer updated his system on the 22nd of December, and went through the disable password + reboot procedure several times since then and still has the issue.

I checked with him how he does the, by instructions suggested-, reboot, and he does so by pressing the top left and right buttons; which does a hard reset, no clean reboot.

Also he made a video; I'm hundred percent sure he really pressed the Disable password option; and last time he did that and did not reboot; and indeed the file is there; hinting that it gets lost during the reboot; rather than that there is an issue creating the file.

System S: https://vrm.victronenergy.com/installation/49974/diagnostics.
System M: https://vrm.victronenergy.com/admin/50275/diagnostics

Both are CCGX-es, he doesn't have this same problem on Octos and Venus-es he installed recently, (which makes sense, on those Remote Console is always enabled & at boot that file is created when it doesn't exist).

System S log, showing a few ubi fs recovery statements (but that might be quite common for many ccgx-es):

```
Dec 29 15:05:50 ccgx daemon.info haveged: haveged: ver: 1.9.4; arch: generic; vend: ; build: (gcc 7.3.0 CTV); collect: 128K
Dec 29 15:05:50 ccgx daemon.info haveged: haveged: cpu: (VC); data: 16K (D); inst: 16K (D); idx: 10/40; sz: 15404/71972
Dec 29 15:05:50 ccgx daemon.info haveged: haveged: tot tests(BA8): A:1/1 B:1/1 continuous tests(B):  last entropy estimate 7.9953
Dec 29 15:05:50 ccgx daemon.info haveged: haveged: fills: 0, generated: 0 
Dec 29 15:05:51 ccgx daemon.info connmand[551]: ntp: time slew +2.667351 s
Dec 29 15:05:56 ccgx user.notice kernel: [   64.700251] random: crng init done
Dec 29 15:20:59 ccgx syslog.info syslogd started: BusyBox v1.24.1
Dec 29 15:20:59 ccgx user.notice kernel: klogd started: BusyBox v1.24.1 (2019-12-20 17:30:18 UTC)
Dec 29 15:20:59 ccgx user.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Dec 29 15:20:59 ccgx user.notice kernel: [    0.000000] Linux version 4.19.74 (oe-user@oe-host) (gcc version 7.3.0 (GCC)) #1 Fri Dec 20 17:59:38 UTC 2019
Dec 29 15:20:59 ccgx user.info kernel: [    0.000000] CPU: ARMv7 Processor [411fc087] revision 7 (ARMv7), cr=10c5387d
Dec 29 15:20:59 ccgx user.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Dec 29 15:20:59 ccgx user.info kernel: [    0.000000] OF: fdt: Machine model: Color Control GX
Dec 29 15:20:59 ccgx user.info kernel: [    0.000000] Memory policy: Data cache writeback
Dec 29 15:20:59 ccgx user.info kernel: [    0.000000] cma: Reserved 16 MiB at 0x8e800000
Dec 29 15:20:59 ccgx user.debug kernel: [    0.000000] On node 0 totalpages: 65024
Dec 29 15:20:59 ccgx user.debug kernel: [    0.000000]   Normal zone: 572 pages used for memmap
Dec 29 15:20:59 ccgx user.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Dec 29 15:20:59 ccgx user.debug kernel: [    0.000000]   Normal zone: 65024 pages, LIFO batch:15
Dec 29 15:20:59 ccgx user.info kernel: [    0.000000] CPU: All CPU(s) started in SVC mode.
Dec 29 15:20:59 ccgx user.info kernel: [    0.000000] AM3517 ES1.1 (l2cache sgx neon)
Dec 29 15:20:59 ccgx user.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x9c/0x484 with crng_init=0
Dec 29 15:20:59 ccgx user.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Dec 29 15:20:59 ccgx user.debug kernel: [    0.000000] pcpu-alloc: [0] 0 
Dec 29 15:20:59 ccgx user.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64452
Dec 29 15:20:59 ccgx user.notice kernel: [    0.000000] Kernel command line:  root=ubi0:rootfs1 rootfstype=ubifs ubi.mtd=12 omapdss.def_disp=lcd omapfb.vram=0:2M@0x8fe00000 fbskip fbcon=vc:64 mtdparts=omap2-nand.0:128k(spl1),128k(spl2),128k(spl3),128k(spl4),1m(u-boot)
Dec 29 15:20:59 ccgx user.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
Dec 29 15:20:59 ccgx user.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
Dec 29 15:20:59 ccgx user.info kernel: [    0.000000] Memory: 224928K/260096K available (6144K kernel code, 479K rwdata, 1988K rodata, 1024K init, 5208K bss, 18784K reserved, 16384K cma-reserved, 0K highmem)
Dec 29 15:20:59 ccgx user.notice kernel: [    0.000000] Virtual kernel memory layout:
Dec 29 15:20:59 ccgx user.notice kernel: [    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
Dec 29 15:20:59 ccgx user.notice kernel: [    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
Dec 29 15:20:59 ccgx user.notice kernel: [    0.000000]     vmalloc : 0xd0000000 - 0xff800000   ( 760 MB)
Dec 29 15:20:59 ccgx user.notice kernel: [    0.000000]     lowmem  : 0xc0000000 - 0xcfe00000   ( 254 MB)
Dec 29 15:20:59 ccgx user.notice kernel: [    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
Dec 29 15:20:59 ccgx user.notice kernel: [    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
Dec 29 15:20:59 ccgx user.notice kernel: [    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (7136 kB)
Dec 29 15:20:59 ccgx user.notice kernel: [    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
Dec 29 15:20:59 ccgx user.notice kernel: [    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 480 kB)
Dec 29 15:20:59 ccgx user.notice kernel: [    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   (5209 kB)
Dec 29 15:20:59 ccgx user.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Dec 29 15:20:59 ccgx user.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Dec 29 15:20:59 ccgx user.info kernel: [    0.000000] IRQ: Found an INTC at 0x(ptrval) (revision 4.0) with 96 interrupts
Dec 29 15:20:59 ccgx user.info kernel: [    0.000000] Clocking rate (Crystal/Core/MPU): 26.0/332/600 MHz
Dec 29 15:20:59 ccgx user.info kernel: [    0.000000] OMAP clockevent source: timer2 at 13000000 Hz
Dec 29 15:20:59 ccgx user.info kernel: [    0.000030] sched_clock: 32 bits at 13MHz, resolution 76ns, wraps every 165191050201ns
Dec 29 15:20:59 ccgx user.info kernel: [    0.000076] clocksource: timer1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 147020034397 ns
Dec 29 15:20:59 ccgx user.info kernel: [    0.000171] OMAP clocksource: timer1 at 13000000 Hz
Dec 29 15:20:59 ccgx user.info kernel: [    0.000951] clocksource: 32k_counter: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 58327039986419 ns
Dec 29 15:20:59 ccgx user.info kernel: [    0.000981] OMAP clocksource: 32k_counter at 32768 Hz
Dec 29 15:20:59 ccgx user.info kernel: [    0.001861] Console: colour dummy device 80x30
Dec 29 15:20:59 ccgx user.info kernel: [    0.004903] console [tty0] enabled
Dec 29 15:20:59 ccgx user.warn kernel: [    0.004965] Lock dependency validator: Copyright (c) 2006 Red Hat, Inc., Ingo Molnar
Dec 29 15:20:59 ccgx user.warn kernel: [    0.005073] ... MAX_LOCKDEP_SUBCLASSES:  8
[... 1824 lines cut ...]
Jan  8 07:29:04 ccgx user.notice kernel: [    3.193800] UBIFS (ubi0:2): media format: w4/r0 (latest is w5/r0), UUID 3E1760C0-1AC3-43F6-9690-847CF05C700D, small LPT model
Jan  8 07:29:04 ccgx daemon.info kernel: [    3.529625] udevd[81]: starting version 3.2.2
Jan  8 07:29:04 ccgx user.notice kernel: [    3.545553] random: udevd: uninitialized urandom read (16 bytes read)
Jan  8 07:29:04 ccgx user.notice kernel: [    3.549097] random: udevd: uninitialized urandom read (16 bytes read)
Jan  8 07:29:04 ccgx user.notice kernel: [    3.549427] random: udevd: uninitialized urandom read (16 bytes read)
Jan  8 07:29:04 ccgx daemon.info kernel: [    3.610657] udevd[82]: starting eudev-3.2.2
Jan  8 07:29:04 ccgx user.notice kernel: [    8.683119] UBIFS (ubi0:0): background thread "ubifs_bgt0_0" started, PID 292
Jan  8 07:29:04 ccgx user.warn kernel: [   10.420219] urandom_read: 3 callbacks suppressed
Jan  8 07:29:04 ccgx user.notice kernel: [   10.420239] random: dd: uninitialized urandom read (512 bytes read)
Jan  8 07:29:04 ccgx user.crit kernel: [   10.795756] watchdog: watchdog0: watchdog did not stop!
[... cut ...]
Jan  8 07:29:04 ccgx user.info kernel: [   12.863094] Bluetooth: HCI socket layer initialized
Jan  8 07:29:04 ccgx user.info kernel: [   12.863225] Bluetooth: L2CAP socket layer initialized
Jan  8 07:29:04 ccgx daemon.info avahi-daemon[529]: Found user 'avahi' (UID 998) and group 'avahi' (GID 997).
Jan  8 07:29:04 ccgx daemon.info avahi-daemon[529]: Successfully dropped root privileges.
Jan  8 07:29:04 ccgx daemon.info avahi-daemon[529]: avahi-daemon 0.6.32 starting up.
Jan  8 07:29:04 ccgx daemon.info avahi-daemon[529]: Suc

mpvader, 2020-01-08:
Log file some the other system, (M):

```
Dec 30 02:48:12 ccgx cron.info CROND[5929]: (root) CMDOUT (installed: 20191220175605 v2.42)
Dec 30 02:48:12 ccgx cron.info CROND[5929]: (root) CMDOUT (available: 20191220175605 v2.42)
Dec 30 02:48:12 ccgx cron.info CROND[5929]: (root) CMDOUT (No newer version available, exit.)
Dec 30 03:23:22 ccgx daemon.info connmand[553]: ntp: time slew +0.013343 s
Dec 30 05:23:22 ccgx daemon.info connmand[553]: ntp: time slew -0.031576 s
Dec 30 07:23:22 ccgx daemon.info connmand[553]: ntp: time slew +0.013283 s
Dec 30 07:49:41 ccgx syslog.info syslogd started: BusyBox v1.24.1
Dec 30 07:49:41 ccgx daemon.info bluetoothd[508]: Bluetooth management interface 1.14 initialized
Dec 30 07:49:41 ccgx user.notice kernel: klogd started: BusyBox v1.24.1 (2019-12-20 17:30:18 UTC)
Dec 30 07:49:41 ccgx user.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Dec 30 07:49:41 ccgx user.notice kernel: [    0.000000] Linux version 4.19.74 (oe-user@oe-host) (gcc version 7.3.0 (GCC)) #1 Fri Dec 20 17:59:38 UTC 2019
Dec 30 07:49:41 ccgx user.info kernel: [    0.000000] CPU: ARMv7 Processor [411fc087] revision 7 (ARMv7), cr=10c5387d
Dec 30 07:49:41 ccgx user.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Dec 30 07:49:41 ccgx user.info kernel: [    0.000000] OF: fdt: Machine model: Color Control GX
Dec 30 07:49:41 ccgx user.info kernel: [    0.000000] Memory policy: Data cache writeback
Dec 30 07:49:41 ccgx user.info kernel: [    0.000000] cma: Reserved 16 MiB at 0x8e800000
Dec 30 07:49:41 ccgx user.debug kernel: [    0.000000] On node 0 totalpages: 65024
Dec 30 07:49:41 ccgx user.debug kernel: [    0.000000]   Normal zone: 572 pages used for memmap
Dec 30 07:49:41 ccgx user.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Dec 30 07:49:41 ccgx user.debug kernel: [    0.000000]   Normal zone: 65024 pages, LIFO batch:15
Dec 30 07:49:41 ccgx user.info kernel: [    0.000000] CPU: All CPU(s) started in SVC mode.
Dec 30 07:49:41 ccgx user.info kernel: [    0.000000] AM3517 ES1.1 (l2cache sgx neon)
Dec 30 07:49:41 ccgx user.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x9c/0x484 with crng_init=0
Dec 30 07:49:41 ccgx user.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Dec 30 07:49:41 ccgx user.debug kernel: [    0.000000] pcpu-alloc: [0] 0 
Dec 30 07:49:41 ccgx user.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64452
Dec 30 07:49:41 ccgx user.notice kernel: [    0.000000] Kernel command line:  root=ubi0:rootfs2 rootfstype=ubifs ubi.mtd=12 omapdss.def_disp=lcd omapfb.vram=0:2M@0x8fe00000 fbskip fbcon=vc:64 mtdparts=omap2-nand.0:128k(spl1),128k(spl2),128k(spl3),128k(spl4),1m(u-boot)
Dec 30 07:49:41 ccgx user.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
Dec 30 07:49:41 ccgx user.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
Dec 30 07:49:41 ccgx user.info kernel: [    0.000000] Memory: 224928K/260096K available (6144K kernel code, 479K rwdata, 1988K rodata, 1024K init, 5208K bss, 18784K reserved, 16384K cma-reserved, 0K highmem)
Dec 30 07:49:41 ccgx user.notice kernel: [    0.000000] Virtual kernel memory layout:
Dec 30 07:49:41 ccgx user.notice kernel: [    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
Dec 30 07:49:41 ccgx user.notice kernel: [    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
Dec 30 07:49:41 ccgx user.notice kernel: [    0.000000]     vmalloc : 0xd0000000 - 0xff800000   ( 760 MB)
Dec 30 07:49:41 ccgx user.notice kernel: [    0.000000]     lowmem  : 0xc0000000 - 0xcfe00000   ( 254 MB)
Dec 30 07:49:41 ccgx user.notice kernel: [    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
Dec 30 07:49:41 ccgx user.notice kernel: [    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
Dec 30 07:49:41 ccgx user.notice kernel: [    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (7136 kB)
Dec 30 07:49:41 ccgx user.notice kernel: [    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
Dec 30 07:49:41 ccgx user.notice kernel: [    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 480 kB)
Dec 30 07:49:41 ccgx user.notice kernel: [    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   (5209 kB)
Dec 30 07:49:41 ccgx user.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Dec 30 07:49:41 ccgx user.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Dec 30 07:49:41 ccgx user.info kernel: [    0.000000] IRQ: Found an INTC at 0x(ptrval) (revision 4.0) with 96 interrupts
Dec 30 07:49:41 ccgx user.info kernel: [    0.000000] Clocking rate (Crystal/Core/MPU): 26.0/332/600 MHz
Dec 30 07:49:41 ccgx user.info kernel: [    0.000000] OMAP clockevent source: timer2 at 13000000 Hz
Dec 30 07:49:41 ccgx user.info kernel: [    0.000029] sched_clock: 32 bits at 13MHz, resolution 76ns, wraps every 165191050201ns
Dec 30 07:49:41 ccgx user.info kernel: [    0.000076] clocksource: timer1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 147020034397 ns
Dec 30 07:49:41 ccgx user.info kernel: [    0.000175] OMAP clocksource: timer1 at 13000000 Hz
Dec 30 07:49:41 ccgx user.info kernel: [    0.000962] clocksource: 32k_counter: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 58327039986419 ns
[... cut ...]
Dec 30 09:24:46 ccgx daemon.notice wpa_supplicant[606]: wlan0: CTRL-EVENT-SIGNAL-CHANGE above=1 signal=-35 noise=9999 txrate=65000
Dec 30 09:24:47 ccgx daemon.notice wpa_supplicant[606]: wlan0: CTRL-EVENT-SIGNAL-CHANGE above=1 signal=-29 noise=9999 txrate=65000
Dec 30 09:24:51 ccgx daemon.notice wpa_supplicant[606]: wlan0: CTRL-EVENT-SIGNAL-CHANGE above=0 signal=-57 noise=9999 txrate=65000
Dec 30 09:24:51 ccgx daemon.notice wpa_supplicant[606]: wlan0: CTRL-EVENT-SIGNAL-CHANGE above=0 signal=-61 noise=9999 txrate=65000
Dec 3

mpvader, 2020-01-08:
@mans, your thoughts?

Wholistic, 2020-01-08:
In my opinion remote console should be enabled by default on all GX devices.
It is secure.

mpvader, 2020-01-08:
For VRM yes; (bold statement though, "its secure" without doing pentesting yourself ;-) ), but Remote Console on LAN is not secure.

mpvader, 2020-01-08:
but noted; though lets not make a long discussion out of this here and now. Regardless of enabling and not enabling by default, still the source of this issue needs to be found. thanks.

closed 2020-01-16 by jhofstee

Wholistic, 2020-02-20:
Possibly related?

Perhaps a Firmware update disabled remote console?

https://community.victronenergy.com/questions/42525/easysolar-ii-gx-remote-console-disconnected.html

mpvader, 2020-02-24:
no thats not related @Wholistic