victronenergy/venus #579: Cerbo: digital inputs dont work

mpvader, 2020-01-08:
the alarm mode configures the gpio pins as edge-triggered interrupts. On the A20 (and other Allwinner chips) only some of the pins support this. the ones someone picked for the digital inputs on the cerbo don't.

short of changing the hardware, we have two options:
1. fall back to periodic polling in the userspace handler
2. hack the kernel driver to poll the pins and emulate interrupts

mpvader, 2020-01-08:
Decided after discussing with Mans:

We’ll modify dbus_digitalinputs to poll. Once a second should be often enough. By doing so on top of the event handling thats already there we (a) have no need to make exceptions and (b) also add a fallback for the (unlikely) case that a rise or fall is missed somewhere in the chain.

Your thoughts @izak ?

We then accept that pulse counting doesn't work on the Cerbo Gx for now; pending a hardware change (if possible, wrt available pins & routing. Mans & Unipoin will work on that).

This is urgent; ie. done & tested this week would be perfect.

RobDuthie, 2020-01-08:
Just a small note that the pulse counting works OK on the Raspi hardware as well as normal inputs high and low. If someone is interested on that function.
I have being testing a Raspi Hat with isolated DIO inputs  to test the full functions of the DIO stage.
Would any changes to the Cerbo GX code effect the raspi coding for this function to work?
Regards
Rob D
NZ

mpvader, 2020-01-08:
Hi Rob, if anything, it will make it better: the digital inputs can then also be assigned to inputs not supporting edge detection aka interrupts.

closed 2020-01-09 by mpvader