victronenergy/venus #584: Decommission Pubnub

mpvader, 2020-01-12:
We were / are still paying for Pubnub; though its no longer functional:

It was only used for remote firmware updates & remote veconfigure; from back in the days that we did that with VE Power Setup. And the currently available VE Power Setup version no longer offers those functions. Trying to use the old features shows this notice:

[image]

Before stopping Pubnub entirely; we need to make sure that the old Venus devices out there won't go all 'crazy' (ie 100% cpu load or some similar thing) if we do so.

Wiebe, do you have time to look into that the coming days?

wiebeytec, 2020-01-14:
I tested with a Venus 1.74. With invalid keys, it fails with:

`ERROR:__main__:Error sending to pubnub :HTTPSConnectionPool(host='pubsub.pubnub.com', port=443): Read timed out.`

Interestingly, when I block port 443 and start vrmpubnub, it doesn't retry after removing the block. The process seems to stop doing anything. 

So:

- No CPU use
- Static RAM use
- No constant error message to the log

I'll keep an eye on it some more today.

wiebeytec, 2020-01-21:
I also monitored traffic to Pubnub, and there is no increase, from repeated retries, for instance. In fact, there is no traffic at all anymore.

I suggest we disable the keys.

mpvader, 2020-01-22:
Ok go for it

wiebeytec, 2020-02-07:
Keys removed and subscription stopped.

closed 2020-02-07 by wiebeytec