victronenergy/venus #598: Inverter RS: fix firmware update via vrm

mpvader, 2020-01-31:
Fails for both VE.Direct and VE.Can. As tested by Roy

And double check that the optimal (= highest) baud rates are being used. Roy has as an action point to test what the maximum baudrate is with a max length (10 meter) VE.Direct cable.

ThiemoVanEngelen, 2020-02-03:
The VE.Direct update fails due to a timeout of 300 seconds and this is just a bit to short. The reason it takes so long is due to the baudrate being 19k2. So there are a number of things that can be improved:

- Use a higher baudrate. As this might not work when people are using longer cables, a backoff algorithm might be a good idea: Start at 460k8, when that fails try 115k2 and finally try 19k2. The xup file specifies which baud rates are supported so a selection from these can be made.
- Make the timeout adaptive for the amount of data (and the baudrate). mqtt-rpc could look at the file (actual data or number of characters) and use a corresponding timeout. When the baudrate is changed, this should also be taken into  account.
- Reset when the timeout when progress is reported. This can be done in 2 ways, both of which should be tested with the update tooling if they behave properly for this behavior and keep reporting progress when they are caught in a loop:
  - Any xml line reported is progress and the timeout is reset
  - Parse the content of the reported xml and only when there is real progress (= the percentage changes) is the timeout reset

ThiemoVanEngelen, 2020-02-03:
The remote update when the inverter is connected via VE.CAN is not possible because the vuppable bit in VE_REG_PRODUCT_ID is not set. This results in an response with `updatable="False"` when scanning devices. Because of this, VRM thinks it cannot be updated.

rvanzundert, 2020-02-03:
Set the VE_PROD_ID_F_VUP_SUPPORT bit in the product id in v01.00-beta-35. After this change the unit could be updated using the vrm portal.

rvanzundert, 2020-02-06:
Full-update of the Inverter RS (used 'time' on command-line to measure the duration):

xupc @250000: 45s
xupd @460800: 27s
xupd @230400: 43s
xupd @115200: 1m15s
xupd @57600: 2m22s
xupd @38400: 3m27s
xupd @19200: 6m46s

xupd tested on internal ve.direct port on the ccgx

About ve.direct cable length:
- Toby found in earlier tests that 10m cable worked at 57k6 and did not work at 115k2 on his universal control board (applies to the Skylla-s and the Multi he is working on), this test was done with a usb ve.direct dongle on a pc with a 10m cable attached to it.
- A quick check on the ve.can mppt chargers indicates that the same restriction apply that Toby found, using 10m cable we cannot go above 57k6.
- I ran all tests on a 10m cable and did not find issues with the ve.direct update on the Inverter RS.

mpvader, 2020-03-31:
As suggested by Thiemo, a quick fix would be to, specifically for the Inverter RS, change mqtt-rpc to use 115k2.

And then other changes can be made later.

mpvader, 2020-04-17:
Hi @ThiemoVanEngelen at what speeds does VictronConnect work? (when connected with a USB-cable)

ThiemoVanEngelen, 2020-04-19:
460k8 if the device supports it. A reason to use "only" 115k2 for Venus is that there is more uncertainty with respect to the cabling (although Roy did test with a 10m cable) and it is a standalone process that the user does not have to wait for so I expect that it is less of an issue when it takes a bit longer.

mpvader, 2020-04-20:
Thiemo already implemented the quick-fix/work around in mqtt-rpc to use 115k2 for the Inverter RS:

https://github.com/victronenergy/mqtt-rpc/commit/7e421fc3d048d6ecaf372600d03b4fb7a0ef9dd2

We still have some time, so next step is that Wiebren will design the final solution first. And then we'll see if we need the quick-fix or can just make the proper solution in one go.

mpvader, 2021-02-09:
solved a long time ago, either per v2.60 and if not there then per v2.62

closed 2021-02-09 by mpvader