victronenergy/venus #599: Add a driver & gui qml page for the car charging station

mpvader, 2020-02-04:
CCS = Car Charging Station

Required functionality:

- auto detect it (on the Ethernet or Wifi network)
- read values (ModbusTCP) and publish on D-Bus
- have an entry in the Device list menu in the gui; showing the values
- controls:
  - Change the mode (5009) between 0-manual and 1-auto
  - Change the charging current (5016) between 6 and 32 Amps

For later:
- add it to the overview page as part of the system.

Tech implementation:

- D-Bus service name: com.victronenergy.carchargingstation (anyone with a better idea can comment :-) ).
- for the list of modbus parameters, see below. Can easily be changed, see people in the Slack Channel.

[image]

mpvader, 2020-11-24:
Included in v2.62~2

closed 2020-11-24 by mpvader

c4software, 2023-07-29:
Hi,

Sorry to resurect this old issue, but, I have a silly question… If I implement this modbus register with my own charging station (ESP32 and VE-Tronic), my device should be available on my venus installation ?

PS: I know it's not the best place to ask. But I don't know where to ask this kind of specific question (related to specific register).

Best regards,