victronenergy/venus #600: Add remote firmware updates of the car charging station

mpvader, 2020-02-04:
What needs to be done is adding it into mqtt-rpc.py, Lucian and Nikolay have already tested proper working of updating the device, doing a http call using curl on a GX Device, see Slack.

izak, 2020-08-20:
mqtt-rpc has been extended (presently it lives on a branch) to provide a more generic device listing as part of the `device-list` command.

In addition to the existing `<duplist>` and `<vuplist>` sections, it will have a `<devices>` block as follows:

```
<devicelist>
    <vuprun version="2.1.1">...</vuprun>
    <duprun version="1.0.21">...</duprun>
    <devices>
        <device serial="BADBADBEEF00" version="1.1.15.24" versionAsDecimal="16846616" 
            description="EV Charging Station" id="49188" type="http"
            connection="http://Victron-EV-Charge-Station.local./firmware.bin"/>
    </devices>
</devicelist>
```

Each <device> has the following attributes:

 * serial: The serial number of the device
 * version: A textual representation of the current firmware version.
 * versionAsDecimal: A numeric value that is used by VRM to compare and find new firmware.
 * description: Textual identification of the product.
 * id: The ProductId of the product.
 * type: How the product updates. Valid values include `can`, `direct`, and `http`.
 * connection: How to connect to the product to do an update. For a CAN-connected
   device this will contain the nad. For VE.Direct devices it will contain the
   serial port. For http updates, it will contain the URL.

`type` and `connection` is used to call `device-update` later, and wil be
passed back as the `connection_type` and `connection_id` parameters.

AlexImb, 2022-02-25:
This was deployed in VRM on December 1st. Unassigning myself.